Kitware/VeloView #5: CMAKE doesn't enable PNG project

neonemrod, 2015-07-23:
Hello:

I am using Ubuntu 14.04, Kernel 3.16.0-41-generic in order to build Veloview. 

I'm trying to build the source but I get this error when trying to do cmake (I think is a problem with my cmakes files or something is missing in my git clone):

```andres@andres-desktop:~/build$ cmake -DENABLE_veloview:BOOL=ON ../veloview/Superbuild
-- Enabling qt since needed by: paraview;pythonqt;veloview
-- Enabling png since needed by: python
CMake Error at CMake/ParaViewModules.cmake:163 (set_property):
  set_property could not find CACHE variable ENABLE_png.  Perhaps it has not
  yet been created.
Call Stack (most recent call first):
 CMakeLists.txt:120 (process_dependencies)

-- Enabling python since needed by: paraview;pythonqt
-- Enabling pythonqt since needed by: paraview
-- Enabling paraview since needed by: veloview
-- Enabling pcap since needed by: veloview
-- Enabling boost since needed by: liblas;veloview
-- Enabling eigen since needed by: veloview
-- Enabling liblas since needed by: veloview
-- Enabling veloview as requested.
-- PROJECTS_ENABLED qt;png;python;pythonqt;paraview;pcap;boost;eigen;liblas;veloview
CMake Error at CMake/ParaViewModules.cmake:188 (include):
  include could not find load file:

    png
Call Stack (most recent call first):
CMakeLists.txt:120 (process_dependencies)```

I tried to modify CMakelist.txt to avoid png but it doesn't have any effect in the compilation or even create another png.cmake, but nothing has worked. I haven't could figure it out how to solve it. 

Can you give me any advice to solve this error, please?

Thanks in advance for your help.

bhjolly, 2015-10-09:
I have this problem as well, same version of Ubuntu. I tried providing ENABLE_png which eliminated the first CMake error but not the second. Is there a 'png' project missing from github or something?

gcasey, 2015-10-14:
The best solution is to enable USE_SYSTEM_QT for Ubuntu.  I'll update the documentation to indicate that and either add an error message that its required or supply the missing png project.

snehaljain018, 2015-11-06:
Hi gcasey,
I received the same errors as neonemrod while superbuild of VeloView. Have you updated the documentation already? And if possible, where in can I change the USE_SYSTEM_QT option for Ubuntu 14.04? I have searched in the top level and low level CMakeLists.txt, without finding anything. Apologies if my query is very amateurish, I am still in the process of learning with a long way to go.

yzrobot, 2015-11-12:
I have the same problem,  enabling USE_SYSTEM_qt can actually solve it. (using ccmake or cmake -DUSE_SYSTEM_qt=ON).

Probably you should upgrade your cmake to 3.2:
sudo apt-get install software-properties-common
sudo add-apt-repository ppa:george-edison55/cmake-3.x
sudo apt-get update
sudo apt-get install --reinstall cmake

timthirion, 2016-03-24:
Hi @neonemrod, @bhjolly, @snehaljain018, & @yzrobot, 

If you're still having build issues, please try the instructions here: https://github.com/Kitware/VeloView/blob/master/Documentation/VeloView_Developer_Guide.rst#linux-build-instructions

wzijian, 2016-12-07:
I use following command:
cmake -DENABLE_veloview:BOOL=ON -DUSE_SYSTEM_python=ON ../veloview/Superbuild

It goes better and passes

choisangwook, 2018-03-13:
Hello

Below shown it result success?
If it success, where is excutable  veloview?

Thank you

sangwook@sangwook:~/catkin_ws/src/VeloView/build$ pwd
/home/sangwook/catkin_ws/src/VeloView/build
sangwook@sangwook:~/catkin_ws/src/VeloView/build$ cmake -DENABLE_veloview:BOOL=ON -DUSE_SYSTEM_python=ON ../Superbuild
-- Enabling qt since needed by: paraview;pythonqt;veloview
-- Enabling python since needed by: paraview;pythonqt
-- Enabling pythonqt since needed by: paraview
-- Enabling paraview since needed by: veloview
-- Enabling pcap since needed by: veloview
-- Enabling boost since needed by: liblas;veloview
-- Enabling eigen since needed by: veloview
-- Enabling liblas since needed by: veloview
-- Enabling veloview as requested.
-- PROJECTS_ENABLED qt;python;pythonqt;paraview;pcap;boost;eigen;liblas;veloview
-- Determined Source Version : 2.0.0-660-gc52ba90
-- Git version (2.0.0-660-gc52ba90) differs from version in file (3.5.0) at /home/sangwook/catkin_ws/src/VeloView/Superbuild/../version.txt
-- Bundled package name will be: VeloView-2.0.0-660-gc52ba90-20180313-Linux-64bit
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sangwook/catkin_ws/src/VeloView/build
sangwook@sangwook:~/catkin_ws/src/VeloView/build$

PierreGuilbertF, 2018-03-14:
Hello,

The VeloView's executable should be in the folder {BuildPath}/install/bin if you made the install command.
If not, the executable will be located in the folder ${BuildPath}/veloview/src/veloview-build/bin.

In your case, after the make install command the VeloView's executable will be located here : 
/home/sangwook/catkin_ws/src/VeloView/build/install/bin

bastienjacquet, 2018-03-14:
Hello choisangwook,

`cmake -DENABLE_veloview:BOOL=ON -DUSE_SYSTEM_python=ON ../Superbuild`
gives you a configured build environnement.
You then need to run:
`cmake --build .` or `make`
To actually build the executables. You will then find the binaries (including VeloView) in the `install/bin` folder.

Best regards, 
Bastien Jacquet, VeloView Project Leader

choisangwook, 2018-03-15:
Dear Bastien Jacquet
​
Thank you for your reply
I have executed all command but can not find the executable anywhere.
And results of make fail.
​
I developed a mobile communication server in Unix environment using C language.
Main tasks are to develop the Home Location Register(HLR), Home Subscriber Server(HSS), Authentication Center(AC), TCP/IP Router and so on, but GitHub's use of Open Source is a novice.
As I have experienced Github, I felt to have a wide range of uses and very little information on the description.
I think there are steps in everything.
I would appreciate it if you would let me know what to study step by step in order to use this part.
​
sangwook@sangwook:~/catkin_ws/src/VeloView$ pwd
/home/sangwook/catkin_ws/src/VeloView
sangwook@sangwook:~/catkin_ws/src/VeloView$ ls -al
total 880
drwxrwxr-x 28 sangwook sangwook   4096  3월 15 09:34 .
drwxrwxr-x  3 sangwook sangwook   4096  3월 15 09:18 ..
-rw-rw-r--  1 sangwook sangwook    273  3월 13 16:15 .clang-format
drwxrwxr-x  8 sangwook sangwook   4096  3월 13 16:15 .git
-rw-rw-r--  1 sangwook sangwook    105  3월 13 16:15 .gitattributes
-rw-rw-r--  1 sangwook sangwook    558  3월 13 16:15 .gitignore
-rw-rw-r--  1 sangwook sangwook     99  3월 13 16:15 .gitmodules
-rw-rw-r--  1 sangwook sangwook  85007  3월 13 16:15 About.png
-rw-rw-r--  1 sangwook sangwook  17934  3월 13 16:15 About.svg
drwxrwxr-x  2 sangwook sangwook   4096  3월 13 16:15 CMake
-rw-rw-r--  1 sangwook sangwook  16691  3월 13 17:41 CMakeCache.txt
drwxrwxr-x 20 sangwook sangwook   4096  3월 15 09:34 CMakeFiles
-rw-rw-r--  1 sangwook sangwook   6763  3월 13 16:15 CMakeLists.txt
-rw-r--r--  1 sangwook sangwook   3643  3월 15 09:14 CPackConfig.cmake
-rw-r--r--  1 sangwook sangwook   4059  3월 13 17:41 CPackSourceConfig.cmake
drwxrwxr-x  2 sangwook sangwook   4096  3월 13 16:15 Documentation
drwxrwxr-x  2 sangwook sangwook   4096  3월 13 16:15 Icons
-rw-rw-r--  1 sangwook sangwook  11345  3월 13 16:15 LICENSE
-rw-rw-r--  1 sangwook sangwook  12243  3월 15 09:34 Makefile
-rw-rw-r--  1 sangwook sangwook   3600  3월 13 16:15 README.md
-rw-rw-r--  1 sangwook sangwook  65633  3월 13 16:15 Splash.jpg
drwxrwxr-x  4 sangwook sangwook   4096  3월 13 16:15 Superbuild
drwxrwxr-x  2 sangwook sangwook   4096  3월 13 16:15 Testing
drwxrwxr-x  3 sangwook sangwook   4096  3월 13 16:15 Utilities
-rw-rw-r--  1 sangwook sangwook   4329  3월 13 16:15 VeloView.svg
drwxrwxr-x  6 sangwook sangwook   4096  3월 13 16:15 VelodyneHDL
drwxrwxr-x  4 sangwook sangwook   4096  3월 13 16:26 boost
drwxrwxr-x  4 sangwook sangwook   4096  3월 13 17:41 boost-prefix
drwxrwxr-x 16 sangwook sangwook   4096  3월 15 09:31 build
-rw-rw-r--  1 sangwook sangwook   2167  3월 13 17:41 cmake_install.cmake
drwxrwxr-x  2 sangwook sangwook   4096  3월 13 16:26 downloads
drwxrwxr-x  4 sangwook sangwook   4096  3월 13 16:26 eigen
-rw-rw-r--  1 sangwook sangwook   2089  3월 13 16:15 generateTestData.py
drwxrwxr-x  3 sangwook sangwook   4096  3월 15 09:15 install
drwxrwxr-x  4 sangwook sangwook   4096  3월 13 16:26 liblas
-rw-rw-r--  1 sangwook sangwook 112943  3월 13 16:15 logo.icns
-rw-rw-r--  1 sangwook sangwook 226574  3월 13 16:15 logo.ico
-rw-rw-r--  1 sangwook sangwook  28079  3월 13 16:15 logo.png
drwxrwxr-x  4 sangwook sangwook   4096  3월 13 16:26 paraview
drwxrwxr-x  4 sangwook sangwook   4096  3월 13 16:26 pcap
drwxrwxr-x  4 sangwook sangwook   4096  3월 13 16:26 png
-rw-rw-r--  1 sangwook sangwook    857  3월 13 16:26 pv-boost-build.cmake
-rw-rw-r--  1 sangwook sangwook    978  3월 13 16:26 pv-boost-configure.cmake
-rw-rw-r--  1 sangwook sangwook    783  3월 13 16:26 pv-eigen-build.cmake
-rw-rw-r--  1 sangwook sangwook    783  3월 13 16:26 pv-liblas-build.cmake
-rw-rw-r--  1 sangwook sangwook    783  3월 13 16:26 pv-paraview-build.cmake
-rw-rw-r--  1 sangwook sangwook    770  3월 13 16:26 pv-pcap-build.cmake
-rw-rw-r--  1 sangwook sangwook    882  3월 13 16:26 pv-pcap-configure.cmake
-rw-rw-r--  1 sangwook sangwook    783  3월 13 16:26 pv-png-build.cmake
-rw-rw-r--  1 sangwook sangwook    783  3월 13 16:26 pv-python-build.cmake
-rw-rw-r--  1 sangwook sangwook    919  3월 13 16:26 pv-python-configure.cmake
-rw-rw-r--  1 sangwook sangwook    783  3월 13 16:26 pv-pythonqt-build.cmake
-rw-rw-r--  1 sangwook sangwook    783  3월 13 16:26 pv-qt-build.cmake
-rw-rw-r--  1 sangwook sangwook   1434  3월 13 16:26 pv-qt-configure.cmake
[... 168 lines cut ...]
[  1%] Building C object VTK/ThirdParty/jpeg/vtkjpeg/CMakeFiles/vtkjpeg.dir/jidctint.c.o
[  1%] Building C object VTK/ThirdParty/jpeg/vtkjpeg/CMakeFiles/vtkjpeg.dir/jidctred.c.o
[  1%] Building C object VTK/ThirdParty/jpeg/vtkjpeg/CMakeFiles/vtkjpeg.dir/jdcolor.c.o
[  1%] Building C object VTK/ThirdParty/jpeg/vtkjpeg/CMakeFiles/vtkjpeg.dir/jdsample.c.o
[  1%] Building C object VTK/ThirdParty/jpeg/vtkjpeg/CMakeFiles/vtkjpeg.dir/jquant1.c.o
[  1%] Linking CXX executable ../../../bin/vtkParseOGLExt-pv5.1
[  1%] Building C object VTK/ThirdParty/jpeg/vtkjpeg/CMakeFiles/vtkjpeg.dir/jquant2.c.o
[  1%] Building C object VTK/ThirdParty/jpeg/vtkjpeg/CMakeFiles/vtkjpeg.dir/jdmerge.c.o
[  1%] Built target vtkParseOGLExt
[  1%] Linking C shared library ../../../../lib/libvtkjpeg-pv5.1.so
[  1%] Built target vtkjpeg
Makefile:127: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMake Error at /home/sangwook/catkin_ws/src/VeloView/build/pv-paraview-build.cmake:26 (message):
 Failed!!!
​
​
CMakeFiles/paraview.dir/build.make:113: recipe for target 'paraview/src/paraview-stamp/paraview-build' failed
make[2]: *** [paraview/src/paraview-stamp/paraview-build] Error 1
CMakeFiles/Makefile2:479: recipe for target 'CMakeFiles/paraview.dir/all' failed
make[1]: *** [CMakeFiles/paraview.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
​
​

bastienjacquet, 2018-03-15:
Hello choisangwook,

It loks like your system cannot compile Paraview, which is needed for VeloView.
What is your Linux flavor and version ?

Best regards,
Bastien Jacquet, VeloView Project Leader

choisangwook, 2018-03-15:
Hello Bastien Jacquet

sangwook@sangwook:~$ uname -a
Linux sangwook 4.13.0-36-generic #40~16.04.1-Ubuntu SMP Fri Feb 16 23:25:58 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

I am not built a ParaView.
I retry it
Thank you