Kitware/VeloView #6: GPRMC length may not always be 72

rkolka, 2015-11-10:
While Velodyne doc says that GPRMC is 72 bytes (and Veloview follows the doc [1]), it may not be. 

"[NMEA] data itself is just ascii text [...] in one variable length sentence. The data may vary in the amount of precision contained in the message. For example time might be indicated to decimal parts of a second or location may be show with 3 or even 4 digits after the decimal point. Programs that read the data should only use the commas to determine the field boundaries and not depend on column positions." [2]

A particular INS [3] sends $GPRMC like this
$GPRMC,114842.000,A,5924.2146614,N,02443.9940738,E,34.8,251.6,051115,8.2,E,D*3F 
and it comes through in Vlodune position packet.

[1] https://github.com/Kitware/VeloView/blob/12f8ff27581f34f87b06794136a202a45f0e1e73/VelodyneHDL/vtkVelodyneHDLPositionReader.cxx#L178 
[2] http://www.gpsinformation.org/dale/nmea.htm
[3] http://www.advancednavigation.com.au/product/spatial-dual

gcasey, 2015-11-12:
The nmea sentance is parsed using commas as recommended.  We only accept a maximum of 72 bytes because AFAIK thats the space allocated within the packet.  Are you seeing weird behavior?

rkolka, 2015-11-13:
Yes, but following bytes are specified as unused:
The position packet is a 554 byte long UDP packet broadcasted on port 8308. It consists of:
42 byte Ethernet header
198 bytes Unused
4 bytes Timestamp
4 bytes Unused
72 byte NMEA $GPRMC sentence
234 bytes Unused

And in particular case they are used - Velodyne packet contains longer RMC message. 

Behavior is no geolocation available. Because some fields are cut off by Veloview, the length of parsed senctence in words is not 13 nor 14 (vtkVelodyneHDLPositionReader line:428). So a default branch (lonDegGPRMC = 0.0; latDegGPRMC = 0.0; heading = 0.0) is taken. 

as these bytes are unused

rkolka, 2015-11-16:
small sample .pcap with longer RMC messages:
https://www.dropbox.com/s/rbp8uxr9nk9qas6/2015-11-05-13-43-56_Velodyne-VLP-16-Data%20%282-12%29.pcap?dl=0

bastienjacquet, 2017-11-09:
Hello,
The code corresponding to VeloView 3.5.0 binaries has been released.
There is some update in handling NMEA $GPRMC sentences.

Please report if it still create problems (I can't test because your link died).

Best regards,
Bastien Jacquet
VeloView Lead Developer

closed 2017-11-09 by bastienjacquet