Kitware/VeloView #8: Mess things up by not using system provided libraries

victl, 2016-01-07:
I'm on ArchLinux, so everything onboard tends to be newest stable version. That is, by default python is on version 3.5.1 and Qt is on 5.5.1 by now.
when I compile VeloView with the provided Superbuild, the libraries vtk, paraview, pythonqt etc. are all required to compile within VeloView itself, system provided versions were not used.
So you can see where the problems are:
1. old pythonqt require python2, but cmake will find python3 for it, then you'll get a compile error. I fixed this by manually add 'include_directories(/usr/include/python2.7)' in it's CMakeLists.txt.
2. old paraview require python2 (and very likely rely on qt4, since I haven't gone through the python2 issue, so not sure if I'd die on qt4 when the current python issue is solved)
Right now I'm stuck on the second issue, could you please help me out?
PS: I admire Kitware for creating CMake. But they seemed very reluctant to keep up with many libraries like python3 and Qt5. This situation not only apply to VeloView, but also with vtk and paraview. I know there might be some tiny patches on certain old libraries version that hinder them from adopting new libraries. But isn't it a bad idea to live on a aged patched library? I'd say this is a very inactive project, not so far from thinking that Kitware is very inactive either.

timthirion, 2016-03-24:
Hi @victl, can you try the updated Linux build instructions here: https://github.com/Kitware/VeloView/blob/master/Documentation/VeloView_Developer_Guide.rst#linux-build-instructions

You will need to use the `pacman` instead of `apt-get`.

Note that it often takes significant developer effort to upgrade a large codebase when dependencies change major versions. You should be able to install both python 2 and 3 as well as Qt 4 and 5 on the same system without issue.

bastienjacquet, 2017-11-09:
Hello,
The code corresponding to VeloView 3.5.0 binaries has been released.
Current code runs with Python 2, and Qt 4.
Some significant developer efforts are being spent on Qt 5 update.
Best regards,
Bastien Jacquet
VeloView Lead Developer

closed 2017-11-09 by bastienjacquet