Kitware/VeloView #9: https://github.com/Kitware/VeloView.git

ctwu0314, 2016-03-15:
Hello:
I try to build VeloView in Window 7 32bit, Visual Studio C++ 2008 Environment.
but I get some error like as below.
Can anyone know how to building correctly.

Nelson

-----command-----
git clone git://public.kitware.com/VeloView.git
mkdir build
cd build
cmake -DENABLE_veloview:BOOL=ON ../VeloView/Superbuild
	Building program using Visual Studio C++ 2008 (debug mode) 

-----Environment-----
Git version 2.7.2.window.1
CMake 2.8.8 

----- error message-----

// -----------------------
CMake Error at python-stamp/download-python.cmake:6 (file):
  file DOWNLOAD MD5 mismatch
    for file: [C:/build_5/downloads/Python-2.7.3.tgz]
      expected MD5 sum: [2cf641732ac23b18d139be077bd906cd]
        actual MD5 sum: [d41d8cd98f00b204e9800998ecf8427e]
Project : error PRJ0019: 工具傳回錯誤碼"Performing download step (download, verify and extract) for 'python'"

//-----------------------
Project : error PRJ0019: 工具傳回錯誤碼"Performing configure step for 'qt'"

//------------------------
libs\regex\build\has_icu_test.cpp(12) : fatal error C1083: 無法開啟包含檔案: 'unicode/uversion.h': No such file or directory

//------------------------
  error: Qt4 was not found on your system.  You probably need to set the
  QT_QMAKE_EXECUTABLE variable

//------------------------
- Could NOT find Qt4 (missing:  QT_QMAKE_EXECUTABLE QT_MOC_EXECUTABLE QT_RCC_EXECUTABLE QT_UIC_EXECUTABLE QT_INCLUDE_DIR QT_LIBRARY_DIR QT_QTCORE_LIBRARY) 
CMake Error at CMakeLists.txt:70 (message):
error: Qt4 was not found on your system.  You probably need to set the
 QT_QMAKE_EXECUTABLE variable
-- Configuring incomplete, errors occurred!
Project : error PRJ0019: 工具傳回錯誤碼"Performing configure step for 'pythonqt'"

//------------------------
-- Could NOT find Qt4 (missing:  QT_QMAKE_EXECUTABLE QT_MOC_EXECUTABLE QT_RCC_EXECUTABLE QT_UIC_EXECUTABLE QT_INCLUDE_DIR QT_LIBRARY_DIR QT_QTCORE_LIBRARY) (Required is at least version "4.7.0")
CMake Error at CMakeLists.txt:441 (message):
  Qt 4.7.0 or greater not found.  Please set the QT_QMAKE_EXECUTABLE

//------------------------
CMake Error at VTK/CMake/FindPythonLibs.cmake:120 (FILE):
  file STRINGS file "C:/build_5/python/src/python/Include/patchlevel.h"
  cannot be read.

//------------------------
CMake Error at C:/Program Files/CMake 2.8/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:97 (MESSAGE):
  Could NOT find PythonInterp: Found unsuitable version "1.4", but required
  is at least "2" (found C:/build_5/python/src/python/PCbuild/python.exe)

//------------------------
CMake Error at CMakeLists.txt:21 (find_package):
  By not providing "FindParaView.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ParaView",
  but CMake did not find one.

timthirion, 2016-03-15:
Hi Nelson, here are the steps I use to build on Windows:

For Windows, you must use Visual Studio 2008. Looks like you're already using that. You should NOT use any USE_SYSTEM options.  The build will take a long time to complete because it has to build both Qt and ParaView from scratch. On my machine it can easily take half a day.

The steps:
1. From the Start Menu, find the Visual Studio 2008 x64 Cross Tools Command Prompt and start it.
2. Run cmake-gui
3. Select the ninja generator
4. Set the CMAKE_BUILD_TYPE to Release
5. Set ENABLE_veloview to ON
6. Set the NMAKE_PATH variable to C:\jom\jom.exe

Generate, change to build directory, then run `ninja`.

Can you give that a try and let me know what happens?

ctwu0314, 2016-03-15:
hi timthirion:
Thank you for your prompt reply.
I have try your method and only modify NMAKE_PATH variable  to 
C:/Program Files/Microsoft Visual Studio 9.0/VC/bin/nmake.exe
The ninja build about one ~ two hour and show some error messages in the end.

Nelson

----- error message-----

NMAKE : fatal error U1077: '"C:\Program Files\Microsoft Visual Studio 9.0\VC\BIN
\cl.EXE"' : 傳回碼 '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files\Microsoft Visual Studio 9.0\VC\bin
\nmake.exe"' : 傳回碼 '0x2'
Stop.
NMAKE : fatal error U1077: 'cd' : 傳回碼 '0x2'
Stop.
NMAKE : fatal error U1077: 'cd' : 傳回碼 '0x2'
Stop.
ninja: build stopped: subcommand failed.

timthirion, 2016-03-15:
That error message means NMake cannot find your compiler. Make sure you are using the Visual Studio 2008 x64 Cross Tools Command Prompt, instead of the usual Windows command prompt. There should be a shortcut in your Start menu under the Visual Studio folder.

ctwu0314, 2016-03-18:
hi timthirion:
I have builded success in Win 64 bit.

I change OS to Win7 64bit then the build error resolved. (NMAKE : fatal error U1077: '"C:\Program Files\Microsoft Visual Studio 9.0\VC\BIN \cl.EXE"' : 傳回碼 '0x2').

And when I build using ninja in WIn 7 64 bit .
the error happened (platform\DefaultLocalizationStrategy.cpp(327) : error C2001: 常數中包含新行字元).
The solved method is "return WEB_UI_STRING("Look Up “<selection>”", "Look Up context menu item with selected word").replace("<selection>", truncatedStringForLookupMenuItem(selectedString)); 
==> \"<selection>\""
and the veloview build success.

Do you build veloview success in Linux ?
Thank you very much.
 
Nelson

timthirion, 2016-03-18:
I have not built VeloView on Linux but will attempt it soon. I'll leave this issue open for now and update when I know how feasible the Linux build is.

timthirion, 2016-03-24:
Hi @ctwu0314, here are the updated instructions for building on Linux: https://github.com/Kitware/VeloView/blob/master/Documentation/VeloView_Developer_Guide.rst#linux-build-instructions

Since this issue concerns building on Windows, I will close it.

closed 2016-03-24 by timthirion

Mrporkk, 2017-11-30:
hi timthirion,
I have builded success in Win10 64 bit and Nmake  generate.But when I open the Veloview.exe of ...install/bin,It shows the lack of python27.dll.

Then I copy the dll to ...install/bin,but Veloview.exe doesn‘t start and show nothing.

Is it a system problem? Or other?

Mrporkk, 2017-11-30:
hi timthirion,
I have builded success in Win10 64 bit and Nmake generate.But when I open the Veloview.exe of ...install/bin,It shows the lack of python27.dll.

Then I copy the dll to ...install/bin,but Veloview.exe doesn‘t start and shows nothing.

Is it a system problem? Or other?

Mrporkk, 2017-11-30:
Hello:
I have builded success in Win10 64 bit and Nmake generate.But when I open the Veloview.exe of ...install/bin,It shows the lack of python27.dll.

Then I copy the dll to ...install/bin,but Veloview.exe doesn‘t start and shows nothing.

Is it a system problem? Or other?