Kitware/VeloView #10: Can't build using superbuild

vishisht, 2016-03-22:
I followed the documentation but cmake gives errors. Following an earlier issue, I added the -DUSE_SYSTEM_dep flag but make still gives errors for boost, python_qt and paraview. Is there updated documentation for how to compile from source?

Thanks,

timthirion, 2016-03-22:
@vishisht, which OS are you trying to build on?

vishisht, 2016-03-22:
Hello Tim,

I'm using Ubuntu 14.04.4 LTS. 

Thanks,
Vishi

timthirion, 2016-03-22:
I'm currently investigating the Linux build. We've had several people ask about it but haven't published official instructions yet. I'll update you here once I have it building.

vishisht, 2016-03-22:
Thanks. The error that I get with the current configuration is:

CMake Warning at /home/vishi/Software/Velo/build/paraview/src/paraview/CMake/ParaViewPlugins.cmake:981 (message):
  GUI resource files in plugins are no longer supported.  The same
  functionality can be obtained using Hints in the Server Manager xml files.
  See the Major API Changes document for details.
Call Stack (most recent call first):
  VelodyneHDL/CMakeLists.txt:161 (add_paraview_plugin)

-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so  
CMake Error at /home/vishi/Software/Velo/build/paraview/src/paraview/CMake/ParaViewPlugins.cmake:1190 (PYTHON_ADD_MODULE):
  Unknown CMake command "PYTHON_ADD_MODULE".
Call Stack (most recent call first):
  VelodyneHDL/CMakeLists.txt:199 (WRAP_PLUGIN_FOR_PYTHON)

-- Configuring incomplete, errors occurred!

Any ideas? It is done building all the dependencies and is building the velodyneHDL source. I have tried adding the python include files and libraries with no change.

Regards,
Vishi

timthirion, 2016-03-22:
What version of CMake are you running? ("cmake --version")

vishisht, 2016-03-22:
cmake version 2.8.12.2

timthirion, 2016-03-22:
Upgrade to 3.2 and try again:

`sudo apt-get install software-properties-common
sudo add-apt-repository ppa:george-edison55/cmake-3.x
sudo apt-get update
sudo apt-get install --reinstall cmake`

vishisht, 2016-03-22:
Fails on pythonqt after I upgraded to cmake version 3.2.2

/usr/include/python3.4m/tupleobject.h:62:75: note: in definition of macro 'PyTuple_SET_ITEM'
 #define PyTuple_SET_ITEM(op, i, v) (((PyTupleObject *)(op))->ob_item[i] = v)
                                                                           ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp: In member function 'QVariant PythonQt::evalCode(PyObject*, PyObject*)':
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:702:125: error: 'PyString_AS_STRING' was not declared in this scope
       globals = PyObject_GetAttrString(PyImport_ImportModule(PyString_AS_STRING(PyObject_GetAttrString(object, "__module__"))),"__dict__");
                                                                                                                             ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:706:64: error: cannot convert 'PyCodeObject*' to 'PyObject* {aka _object*}' for argument '1' to 'PyObject* PyEval_EvalCode(PyObject*, PyObject*, PyObject*)'
       r = PyEval_EvalCode((PyCodeObject*)pycode, globals , dict);
                                                                ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp: In member function 'QStringList PythonQt::introspectObject(PyObject*, PythonQt::ObjectType)':
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:893:43: error: 'PyString_AsString' was not declared in this scope
           results << PyString_AsString(doc);
                                           ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:921:39: error: 'PyString_AsString' was not declared in this scope
         keystr = PyString_AsString(key);
                                       ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:929:36: error: 'PyClass_Type' was not declared in this scope
             if (value->ob_type == &PyClass_Type || value->ob_type == &PyType_Type) {
                                    ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:934:36: error: 'PyClass_Type' was not declared in this scope
             if (value->ob_type != &PyClass_Type
                                    ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp: In function 'int {anonymous}::custom_system_exit_exception_handler()':
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:1278:20: error: 'Py_FlushLine' was not declared in this scope
   if (Py_FlushLine())
                    ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:1295:24: error: 'PyInt_Check' was not declared in this scope
   if (PyInt_Check(value))
                        ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:1296:39: error: 'PyInt_AsLong' was not declared in this scope
     exitcode = (int)PyInt_AsLong(value);
                                       ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp: In member function 'void PythonQt::initPythonQtModule(bool, const QByteArray&)':
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:1439:72: error: 'Py_InitModule' was not declared in this scope
   _p->_pythonQtModule = Py_InitModule(name.constData(), PythonQtMethods);
                                                                        ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp: In member function 'QString PythonQt::getReturnTypeOfWrappedMethodHelper(const PythonQtObjectPtr&, const QString&, const QString&)':
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:1496:33: error: 'PyClass_Type' was not declared in this scope
   if (methodObject->ob_type == &PyClass_Type || methodObject->ob_type == &PyType_Type) {
                                 ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:1510:54: error: 'PyString_AsString' was not declared in this scope
         className = PyString_AsString(classNameObject);
                                                      ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:1544:51: error: 'PyString_AsString' was not declared in this scope
                 type = QString(PyString_AsString(s)) + "." + type;
                                                   ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp: In member function 'PyObject* PythonQt::helpCalled(PythonQtClassInfo*)':
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:1652:62: error: 'PyString_FromString' was not declared in this scope
     return PyString_FromString(info->help().toLatin1().data());
                                                              ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp: In member function 'bool PythonQtPrivate::isMethodDescriptor(PyObject*) const':
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:1728:28: error: 'PyClass_Check' was not declared in this scope
       !PyClass_Check(object)) {
                            ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp: In member function 'QString PythonQtPrivate::getSignature(PyObject*)':
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:1744:29: error: 'PyClass_Type' was not declared in this scope
[... cut ...]
                                                 ^
make[5]: *** [CMakeFiles/PythonQt.dir/src/PythonQtConversion.cpp.o] Error 1
make[5]: *** [CMakeFiles/PythonQt.dir/src/PythonQt.cpp.o] Error 1
make[4]: *** [CMakeFiles/PythonQt.dir/all] Error 2
make[3]: *** [all] Error 2
CMake Error at /home/vishi/Software/Velo/build/pv-pythonqt-build.cmake:26 (message):
  Failed!!!

make[2]: *** [pythonqt/src/pythonqt-stamp/pythonqt-build] Error 1
make[1]: *** [CMakeFiles/pythonqt.dir/all] Error 2
make: *** [all] Error 2

timthirion, 2016-03-22:
The reason for your error is not immediately coming to my mind. I'll update with the build instructions when I get through it myself. I'm doing this on my own time, so it might be a few days.

vishisht, 2016-03-22:
Thanks much.

timthirion, 2016-03-24:
Hi @vishisht, can you try the updated build instructions located here: https://github.com/Kitware/VeloView/blob/master/Documentation/VeloView_Developer_Guide.rst#linux-build-instructions

Once you're able to build, I will close this issue.

vishisht, 2016-03-25:
Still getting an error at pythonqt build step. Here is the error.

Scanning dependencies of target pythonqt
[ 56%] Creating directories for 'pythonqt'
[ 58%] Performing download step (git clone) for 'pythonqt'
Cloning into 'pythonqt'...
Switched to a new branch 'patched-3'
Branch patched-3 set up to track remote branch patched-3 from origin.
[ 59%] No patch step for 'pythonqt'
[ 61%] Performing update step for 'pythonqt'
Current branch patched-3 is up to date.
[ 62%] Performing configure step for 'pythonqt'
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.4m.so (found version "3.4.3") 
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /usr/bin/qmake (found version "4.8.6") 
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_CXX_FLAGS_DEBUG
    CMAKE_CXX_FLAGS_MINSIZEREL
    CMAKE_CXX_FLAGS_RELEASE
    CMAKE_CXX_FLAGS_RELWITHDEBINFO
    CMAKE_C_FLAGS_DEBUG
    CMAKE_C_FLAGS_MINSIZEREL
    CMAKE_C_FLAGS_RELEASE
    CMAKE_C_FLAGS_RELWITHDEBINFO
    MACOSX_APP_INSTALL_PREFIX

-- Build files have been written to: /home/vishi/Software/Velo/build/pythonqt/src/pythonqt-build
[ 63%] Performing build step for 'pythonqt'
[  4%] [  4%] [  4%] [  4%] [  5%] Generating src/moc_PythonQt.cxx
Generating src/gui/moc_PythonQtScriptingConsole.cxx
Generating src/moc_PythonQtSignalReceiver.cxx
Generating src/moc_PythonQtStdDecorators.cxx
Generating generated_cpp/com_trolltech_qt_uitools/moc_com_trolltech_qt_uitools0.cxx
[  7%] [  8%] [  9%] Generating generated_cpp/com_trolltech_qt_core_builtin/moc_com_trolltech_qt_core_builtin0.cxx
[ 10%] [ 11%] Generating generated_cpp/com_trolltech_qt_gui_builtin/moc_com_trolltech_qt_gui_builtin0.cxx
Generating generated_cpp/com_trolltech_qt_core/moc_com_trolltech_qt_core0.cxx
Generating generated_cpp/com_trolltech_qt_core/moc_com_trolltech_qt_core1.cxx
Generating generated_cpp/com_trolltech_qt_core/moc_com_trolltech_qt_core2.cxx
[ 12%] [ 14%] Generating generated_cpp/com_trolltech_qt_gui/moc_com_trolltech_qt_gui0.cxx
[ 15%] Generating generated_cpp/com_trolltech_qt_gui/moc_com_trolltech_qt_gui1.cxx
Generating generated_cpp/com_trolltech_qt_gui/moc_com_trolltech_qt_gui2.cxx
[ 16%] Generating generated_cpp/com_trolltech_qt_gui/moc_com_trolltech_qt_gui3.cxx
[ 17%] Generating generated_cpp/com_trolltech_qt_gui/moc_com_trolltech_qt_gui4.cxx
[ 18%] [ 20%] Generating generated_cpp/com_trolltech_qt_gui/moc_com_trolltech_qt_gui6.cxx
Generating generated_cpp/com_trolltech_qt_gui/moc_com_trolltech_qt_gui5.cxx
[... 338 lines cut ...]
                                        ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:1822:44: error: 'PyString_AsString' was not declared in this scope
           varkeywords = PyString_AsString(s);
                                            ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:1833:42: error: 'PyString_AsString' was not declared in this scope
           defaults << PyString_AsString(s);
                                          ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp: In static member function 'static PyObject* PythonQtPrivate::wrapMemoryAsBuffer(const void*, Py_ssize_t)':
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:1885:47: error: 'PyBuffer_FromMemory' was not declared in this scope
   return PyBuffer_FromMemory((void*)data, size);
                                               ^
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp: In static member function 'static PyObject* PythonQtPrivate::wrapMemoryAsBuffer(void*, Py_ssize_t)':
/home/vishi/Software/Velo/build/pythonqt/src/pythonqt/src/PythonQt.cpp:1891:49: error: 'PyBuffer_FromReadWriteMemory' was not declared in this scope
   return PyBuffer_FromReadWriteMemory(data, size);
                                                 ^
make[5]: *** [CMakeFiles/PythonQt.dir/src/PythonQtConversion.cpp.o] Error 1
make[5]: *** [CMakeFiles/PythonQt.dir/src/PythonQt.cpp.o] Error 1
make[4]: *** [CMakeFiles/PythonQt.dir/all] Error 2
make[3]: *** [all] Error 2
CMake Error at /home/vishi/Software/Velo/build/pv-pythonqt-build.cmake:26 (message):
  Failed!!!

make[2]: *** [pythonqt/src/pythonqt-stamp/pythonqt-build] Error 1
make[1]: *** [CMakeFiles/pythonqt.dir/all] Error 2
make: *** [all] Error 2

timthirion, 2016-03-25:
It looks like you need python 2 rather than python 3. Could you install that and try again? You may need to uninstall python 3.

vishisht, 2016-03-25:
Thanks for your response. Now I'm able to get to the velodyne compile, but get the error on PYTHON_ADD_MODULE. My cmake version is 3.2.2.

[ 73%] Completed 'paraview'
[ 77%] Built target paraview
[ 88%] Built target pcap
[ 90%] Performing configure step for 'veloview'
CMake Error at CMakeLists.txt:27 (include):
  include called with wrong number of arguments.  include() only takes one
  file.

-- Eigen found (include: /home/vishi/Software/Velo/build/install/include/eigen3)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   thread
--   chrono
--   regex
--   date_time
--   filesystem
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   atomic
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   timer
CMake Warning at /home/vishi/Software/Velo/build/paraview/src/paraview/CMake/ParaViewPlugins.cmake:981 (message):
  GUI resource files in plugins are no longer supported.  The same
  functionality can be obtained using Hints in the Server Manager xml files.
  See the Major API Changes document for details.
Call Stack (most recent call first):
  VelodyneHDL/CMakeLists.txt:161 (add_paraview_plugin)

CMake Error at /home/vishi/Software/Velo/build/paraview/src/paraview/CMake/ParaViewPlugins.cmake:1190 (PYTHON_ADD_MODULE):
  Unknown CMake command "PYTHON_ADD_MODULE".
Call Stack (most recent call first):
  VelodyneHDL/CMakeLists.txt:199 (WRAP_PLUGIN_FOR_PYTHON)

-- Configuring incomplete, errors occurred!
See also "/home/vishi/Software/Velo/build/veloview/src/veloview-build/CMakeFiles/CMakeOutput.log".
See also "/home/vishi/Software/Velo/build/veloview/src/veloview-build/CMakeFiles/CMakeError.log".
make[2]: *** [veloview/src/veloview-stamp/veloview-configure] Error 1
make[1]: *** [CMakeFiles/veloview.dir/all] Error 2
make: *** [all] Error 2

timthirion, 2016-03-28:
This still looks like a CMake version problem. Are you sure you don't have two CMake binaries on your system?

vishisht, 2016-03-30:
Turns out it was a python issue. When I included the python 2.7 paths consistently, I was able to build successfully. But when I run the program, I get a core dump. 
[image]

vishisht, 2016-03-30:
I had Paraview installed separately. I removed it and I don't get this exception. Getting another error now. But should be able to figure it out. Thanks a lot for your help.

timthirion, 2016-03-30:
Happy to help! I'll close this issue but if you get stuck again, feel free to open another.

closed 2016-03-30 by timthirion

nlw0, 2016-12-18:
...So how do I properly configure to use python 2.7?

nlw0, 2016-12-18:
Might help someone else: this is how I configured on my Ubuntu 16.04. Bonus: how to just configure everything from the command line so you don't have to open ccmake, or whatever

`cmake ../VeloView/Superbuild/ -DPYTHON_EXECUTABLE:FILEPATH=/usr/bin/python2.7 -DPYTHON_LIBRARY:FILEPATH=/usr/lib/python2.7/config-x86_64-linux-gnu/libpython2.7.so -DPYTHON_INCLUDE_DIR:FILEPATH=/usr/include/python2.7/  -DENABLE_veloview=ON -DUSE_SYSTEM_boost=ON -DUSE_SYSTEM_python=ON -DUSE_SYSTEM_qt=ON`