Kitware/VeloView #14: Ubuntu 16.04 issues

cmeyers-st, 2016-04-27:
Failure to build pythonqt. On a system with python3 as default or python3 and python2 installed the generic build is picking up python3 header files. I changed PythonQtPythonInclude.h to explicitly include python2.7/Python.h

Seems to work.

timthirion, 2016-04-27:
@cmeyers-st, do you want to put up a pull request with your changes? It might be good to just explicitly support python 2.7 for now, since that's the required version.

cc @bastienjacquet

cmeyers-st, 2016-04-27:
Well, I quickly ran into problems with VTK paraview and haven't been able to fix them, also related to python3. Also I am not sure my fix is the right way, it is possible to restrict cmake to a specific version but I have not been able to make it work as documented.

timthirion, 2016-04-27:
https://github.com/Kitware/VeloView/issues/10

timthirion, 2016-04-27:
See also: https://github.com/Kitware/VeloView/issues/10

@vishisht, any tips for building with python 2 and 3 installed?

vishisht, 2016-04-27:
Hello Tim,

All I had to do was to configure the correct python paths while running ccmake. I did not need to un-install python 3.

Regards,
Vishi

cmeyers-st, 2016-04-27:
Care to provide details on how you configured cmake to only use python2.7?

This has to pass down to paraview because that is what is failing after I fixed the VeloView python wrapper.

vishisht, 2016-04-27:
In the build directory.
Ccmake Veloview/Superbuild

[cid:image003.jpg@01D1A07D.48925200]

It builds fine after I did this, but while running it still complains about paths. So, I had to set some paths for python while running the executable for it to function properly.

bastienjacquet, 2017-11-09:
Closing this since it's resolved.

closed 2017-11-09 by bastienjacquet