Kitware/VeloView #16: make error: undefined reference to `PyVTKClass_vtkImageStencilAlgorithmNew'

bcn2, 2016-06-02:
make stops with error 2, after this one:
[ 44%] Built target vtkFiltersExtractionCS
make[5]: Entering directory '/velo/build/paraview/src/paraview-build'
CMakeFiles/vtkImagingCorePythonD.dir/vtkImageStencilSourcePython.cxx.o: In function `PyVTKClass_vtkImageStencilSourceNew':
vtkImageStencilSourcePython.cxx:(.text+0xbf6): undefined reference to `PyVTKClass_vtkImageStencilAlgorithmNew'
collect2: error: ld returned 1 exit status
make[5]: *** [lib/libvtkImagingCorePython27D-pv4.3.so.1] Error 1
VTK/Wrapping/Python/CMakeFiles/vtkImagingCorePythonD.dir/build.make:1408: recipe for target 'lib/libvtkImagingCorePython27D-pv4.3.so.1' failed
make[5]: Leaving directory '/velo/build/paraview/src/paraview-build'
make[4]: *** [VTK/Wrapping/Python/CMakeFiles/vtkImagingCorePythonD.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:19597: recipe for target 'VTK/Wrapping/Python/CMakeFiles/vtkImagingCorePythonD.dir/all' failed
make[5]: Leaving directory '/velo/build/paraview/src/paraview-build'
...
building on an Olimex ARM A20 Lime2, with armbian Jessie

bcn2, 2016-06-02:
referenced file had 0 Bytes; file deleted and make restarted

closed 2016-06-02 by bcn2

timthirion, 2016-06-02:
Just curious, any idea why you had an empty file? Could be indicative of a different problem.

bcn2, 2016-06-02:
the board is instable, most probably a Jessie problem with that platform. On the olimex web page the recommendation is to go still with wheezy.