Kitware/VeloView #17: VeloView Developer Guide, more Linux packages needed

bcn2, 2016-06-02:
in
1.6 Linux build instructions
please add libqt4-opengl-dev 

We installed it - using Debian Jessie - after getting this error:
CMake Error at /usr/share/cmake-3.0/Modules/FindPackageHandleStandardArgs.cmake:136 (message):
  Could NOT find OpenGL (missing: OPENGL_gl_LIBRARY)
Call Stack (most recent call first):
  /usr/share/cmake-3.0/Modules/FindPackageHandleStandardArgs.cmake:343 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.0/Modules/FindOpenGL.cmake:175 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  VTK/Rendering/OpenGL/CMakeLists.txt:196 (find_package)

timthirion, 2016-06-02:
Did you install graphics drivers previous to building VeloView? Or is this on your machine that has experimental drivers?

I'm happy to add that dependency; just hoping to understand your build environment better.

bcn2, 2016-06-02:
This is on the  Olimex ARM A20 Lime2, with armbian Jessie, Mali device driver (sunxi-mali-r3p0).

closed 2016-06-02 by bcn2