Kitware/VeloView #18: Clarification on Data Acquisition

jbuck167, 2016-07-11:
This probably fits under "Help wanted" 

I am currently using VeloView to collect data from a VLP-16 scanner and exporting the data as a CSV/VTK supported format (I am actually then converting those formats to OBJ and PLY for easier use). I then have created a custom C++ VTK/PCL application that uses the data from the VLP-16. I want to add support for my application to read data directly from the scanner (and convert to a vtk/pcl friendly format). Since you have presumably already implemented this in Veloview, is it possible to point me towards the data acquisition component or files of Veloview? In exchange, I am willing to share my language independent implementation of saving the velodyne data files as (.OBJ) and (.PLY). I also have experience registering and aligning data samples from the Velodyne sensor- if any of those features are of interest to you. I do not currently have a presence of Github as I am an Atlassian Bitbucket fan, but I am experienced with C++, Python, VTK, and Qt-it appears that is what you are using for VeloView.

jbuck167, 2016-07-11:
It looks like "TestSensor.cxx" contains the implementation that I was looking for, specifically void vvPacketSender::pumpPacket() and I assume "vtkVelodyneHDLPositionReader" converts the raw data angle/azimuth to XYZ coordinates? Does Veloview primarily work with the angle/azimuth data?

timthirion, 2016-07-12:
@jbuck167 I believe that is all correct, though it's been awhile since I've read the source code. Does your output look right?

bastienjacquet, 2016-07-12:
Hello @jbuck167,
Thanks for your interest in VeloView!
To answer your questions:
vtkVelodyneHDLPositionReader reads the GPS/IMU packets
vtkVelodyneHDLReader reads the laser firing packets

And yes, Veloview primarily work with the angle/azimuth data. 
As for the GPS data, it only uses position and heading (i.e. it assumes a planar motion). 
[see vtkVelodyneHDLPositionReader.cxx:251]

jbuck167, 2016-07-14:
Another clarification: vvPacketSender is used to read from a pcap file and essentially "spoof" the sensor by sending data on the Velodyne sensor port (2368)? vtkVelodyneHDLReader then reads the data as if it came from the sensor and not a previously captured file?

bastienjacquet, 2016-12-30:
> Further clarification: vvPacketSender is used to read from a pcap file and essentially "spoof" the sensor by sending data on the Velodyne sensor port (2368)? 
Exactly.

> vtkVelodyneHDLReader then reads the data as if it came from the sensor and not a previously captured file?
Exactly.
TestSensor.cxx does this within the same program.
Please note that on some platforms two different machines are needed because the way we send packets to ourself does not work. In that case, PacketFileSender can be used on another machine.

Best,
Bastien,
VeloView Lead Developer

closed 2017-09-20 by bastienjacquet