Kitware/VeloView #20: Ubuntu 16.04 CMake CMake/ParaViewModules.cmake:188

maym86, 2016-09-23:
Getting this error:

```
CMake Error at CMake/ParaViewModules.cmake:188 (include):
  include could not find load file:

    png
Call Stack (most recent call first):
  CMakeLists.txt:124 (process_dependencies)
```

This is how I am running cmake

cmake -DENABLE_veloview:BOOL=ON -DUSE_SYSTEM_python:BOOL=ON -DUSE_SYSTEM_boost:BOOL=ON -DUSE_SYSTEM_qt:BOOL=ON -DUSE_SYSTEM_pythonqt:BOOL=ON ../Superbuild

Thanks
Mike

timthirion, 2016-09-26:
Your build directory is inside the source directory? If so, that may cause problems. I recommend putting the build directory in a wholly separate place in your filesystem.

Second, I assume you've installed all the `apt-get` dependencies, as outlined in the Developer's Guide?

wzijian, 2016-12-07:
I use this and it's OK:

cmake -DENABLE_veloview:BOOL=ON -DUSE_SYSTEM_python=ON ../veloview/Superbuild

maym86, 2016-12-07:
Ended up using the install_veloview.sh script attached with a modifiled qt.cmake and CMakeLists.txt

[CMakeLists.txt](https://github.com/Kitware/VeloView/files/637994/CMakeLists.txt)

closed 2016-12-07 by maym86