Kitware/VeloView #22: Build fails on Ubuntu 14.04 (following build instructions)

Chambana, 2016-10-12:
After several attempts, I cannot build VeloView on Ubuntu 14.04 despite following the instructions.  I've provided a detailed account below of my approach and the results.  Any help would be appreciated.  

Install environment:
	Ubuntu 14.04
	cmake version 3.2.2
	QMake version 2.01a
	Using Qt version 4.8.6 in /usr/lib/x86_64-linux-gnu
	Python 2.7.6
	Python 3.4.3

Cloned into ./VeloView using:
	git clone git://public.kitware.com/VeloView.git

Installed dependencies with apt-get:
	sudo apt-get install git cmake-curses-gui build-essential libboost-all-dev libxt-dev libbz2-dev libqt4-dev qt4-default qt4-dev-tools zlib1g-dev

	build-essential is already the newest version.
	libbz2-dev is already the newest version.
	libxt-dev is already the newest version.
	zlib1g-dev is already the newest version.
	libboost-all-dev is already the newest version.
	git is already the newest version.
	libqt4-dev is already the newest version.
	qt4-default is already the newest version.
	qt4-dev-tools is already the newest version.
	cmake-curses-gui is already the newest version.

(Re-)installed updated cmake:
	sudo apt-get install software-properties-common
	sudo add-apt-repository ppa:george-edison55/cmake-3.x
	sudo apt-get update
	sudo apt-get install --reinstall cmake

Installed additional dependencies per the instructions:
	sudo apt-get install flex byacc 

From the newly created veloview build directory:
	ccmake ~/VeloView/Superbuild/
	hit 'c' to Configure once inside cmake

CCMAKE displays these default options:
	 CMAKE_BUILD_TYPE                *
	 CMAKE_INSTALL_PREFIX            */usr/local
	 ENABLE_boost                    *OFF
	 ENABLE_eigen                    *OFF
	 ENABLE_liblas                   *OFF
	 ENABLE_paraview                 *OFF
	 ENABLE_pcap                     *OFF
	 ENABLE_python                   *OFF
	 ENABLE_pythonqt                 *OFF
	 ENABLE_qt                       *OFF
	 ENABLE_veloview                 *OFF
	 qt_EXTRA_CONFIGURATION_OPTIONS  *       

The build instructions say to change the following to ON:
	ENABLE_veloview
	USE_SYSTEM_boost 
	USE_SYSTEM_python
	USE_SYSTEM_qt

However, the options USE_SYSTEM_boost, USE_SYSTEM_python, and USE_SYSTEM_qt are not visible.  First set ENABLE_boost, ENABLE_python, and ENABLE_qt to ON, then the 'USE_SYSTEM' options will be visible.  

The following error appears:
	 CMake Error at CMake/ParaViewModules.cmake:163 (set_property):
	   set_property could not find CACHE variable ENABLE_png.  Perhaps it has not
	   yet been created.
	 Call Stack (most recent call first):
	   CMakeLists.txt:120 (process_dependencies)
	 CMake Error at CMake/ParaViewModules.cmake:188 (include):
	   include could not find load file:
[... 398 lines cut ...]
	           varargs = PyString_AsString(s);
	                                        ^
	/home/drew/veloview_build_directory/pythonqt/src/pythonqt/src/PythonQt.cpp:1822:44: error: 'PyString_AsString' was not declared in this scope
	           varkeywords = PyString_AsString(s);
	                                            ^
	/home/drew/veloview_build_directory/pythonqt/src/pythonqt/src/PythonQt.cpp:1833:42: error: 'PyString_AsString' was not declared in this scope
	           defaults << PyString_AsString(s);
	                                          ^
	/home/drew/veloview_build_directory/pythonqt/src/pythonqt/src/PythonQt.cpp: In static member function 'static PyObject* PythonQtPrivate::wrapMemoryAsBuffer(const void*, Py_ssize_t)':
	/home/drew/veloview_build_directory/pythonqt/src/pythonqt/src/PythonQt.cpp:1885:47: error: 'PyBuffer_FromMemory' was not declared in this scope
	   return PyBuffer_FromMemory((void*)data, size);
	                                               ^
	/home/drew/veloview_build_directory/pythonqt/src/pythonqt/src/PythonQt.cpp: In static member function 'static PyObject* PythonQtPrivate::wrapMemoryAsBuffer(void*, Py_ssize_t)':
	/home/drew/veloview_build_directory/pythonqt/src/pythonqt/src/PythonQt.cpp:1891:49: error: 'PyBuffer_FromReadWriteMemory' was not declared in this scope
	   return PyBuffer_FromReadWriteMemory(data, size);
	                                                 ^
	make[5]: *** [CMakeFiles/PythonQt.dir/src/PythonQt.cpp.o] Error 1
	make[4]: *** [CMakeFiles/PythonQt.dir/all] Error 2
	make[3]: *** [all] Error 2
	CMake Error at /home/drew/veloview_build_directory/pv-pythonqt-build.cmake:26 (message):
	  Failed!!!

	make[2]: *** [pythonqt/src/pythonqt-stamp/pythonqt-build] Error 1
	make[1]: *** [CMakeFiles/pythonqt.dir/all] Error 2
	make: *** [all] Error 2

timthirion, 2016-10-12:
Remove python 3 and give it another shot.

Chambana, 2016-10-13:
I un-installed Python3 using "apt remove" which, on Ubuntu 14.04, also uninstalled about half my OS, including Ubuntu's Unity desktop manager :)

Reinstalling Unity unfortunately reinstalled Python3 as well.  However, I did make some progress.  I went into ccmake and noticed that, in the advanced options, cmake was directing several options to:
    /usr/bin/python3.4
    /usr/lib/x86_64-linux-gnu/libpython3.4m.so.1
etc, etc

I changed those options to point to python2.7 instead of python3.4.  And now, when running make, pythonQT finished compiling and VeloView was successfully built!

Chambana, 2016-10-13:
A few more issues (now resolved).  

The above procedure got VeloView to load, however, it threw errors that VeloView couldn't import paraview.simple and wouldn't display LIDAR data.  So, I had to point my PYTHONPATH to
     /veloview_build_directory/install/lib/paraview-4.3/site-packages       
     /veloview_build_directory/install/lib/velowview3.1/site-packages

That got paraview.simple to import properly when loading VeloView.  However, it then couldn't import VtkCommonCorePython when loading VeloView:
        
        from vtkCommonCorePython import *
        ImportError: No module named vtkCommonCorePython

So, I added the following to my $PYTHONPATH
       /veloview_build_directory/install/lib/paraview-4.3/site-packages/vtk 

This got VeloView to load without error and display the 3D grid. However, loading a pcap file with LIDAR data threw an error that no Calibration File was loaded (even though I selected HDL32).  

Another user posted his .xml HDL32e calibration file here, which can be loaded manually in VeloView and allow for LIDAR pcap files to be viewed:
     https://dl.dropboxusercontent.com/u/95042389/32db.xml

Chambana, 2016-10-13:
UPDATE:  The Calibration File for the HDL32s is actually included in the velodyne build directory under /install/share

closed 2016-10-13 by Chambana