Kitware/VeloView #23: PythonQT build cmake failed

andrewraharjo, 2016-11-07:
I got an issue building from the source on Ubuntu 14.04.4

Here's the steps I've taken:
1. Clone the source
2. Download all required dependencies
3. Configure ccmake per what it's instructed on the separate build folder (outside from the source directory) CMAKE_BUILD_TYPE = RELEASE
4.  Export QT Select to 4
5. run cmake -DENABLE_veloview:BOOL=ON ../veloview/SuperBuild
6. then I do make

Here 's the error output from make :
`
/home/andrewraharjo/build/pythonqt/src/pythonqt/src/PythonQtImporter.cpp: In static member function ‘static void PythonQtImport::init()’:
/home/andrewraharjo/build/pythonqt/src/pythonqt/src/PythonQtImporter.cpp:798:36: error: ‘Py_InitModule4’ was not declared in this scope
            NULL, PYTHON_API_VERSION);
                                    ^
make[5]: *** [CMakeFiles/PythonQt.dir/src/PythonQtImporter.cpp.o] Error 1
make[5]: *** [CMakeFiles/PythonQt.dir/src/PythonQt.cpp.o] Error 1
make[4]: *** [CMakeFiles/PythonQt.dir/all] Error 2
make[3]: *** [all] Error 2
CMake Error at /home/andrewraharjo/build/pv-pythonqt-build.cmake:26 (message):
  Failed!!!

make[2]: *** [pythonqt/src/pythonqt-stamp/pythonqt-build] Error 1
make[1]: *** [CMakeFiles/pythonqt.dir/all] Error 2
make: *** [all] Error 2
`

andrewraharjo, 2016-11-07:
Oh I just realized the `ccmake` directs to Python3, Python 2.7. It solves the problem when it directs to 2.7, now continuing to do paraview

bastienjacquet, 2016-12-30:
Hello,
I'm glad you fixed the issue.
Best,
Bastien,
VeloView Lead Developer

closed 2017-09-20 by bastienjacquet