Kitware/VeloView #24: build target paraview failed

Grillen80, 2016-12-02:
I'm trying get veloview working under ubutun 16.04. I used all the setting recommend in https://github.com/Kitware/VeloView/blob/master/Documentation/VeloView_Developer_Guide.rst#getting-the-source-code
Unfortunately the build  stops  with error at "Build Target paraview", Makefile:149, make: **[all]
without any detailed information.
Is where a workaround or could anyone help me out?
Thanks, Sven

timthirion, 2016-12-02:
Can you do `make VERBOSE=1` and see if you get any more information?

Grillen80, 2016-12-05:
The VERBOSE argument didn't helped me out. I've got the messages as follows:

[ 91%] Performing build step for 'veloview'
cd /home/ecke/Downloads/VeloView/veloview/src/veloview-build && /usr/bin/cmake -P /home/ecke/Downloads/VeloView/pv-veloview-build.cmake
make[3]: Verzeichnis „/home/ecke/Downloads/VeloView/veloview/src/veloview-build“ wird betreten
make[3]: Verzeichnis „/home/ecke/Downloads/VeloView/veloview/src/veloview-build“ wird verlassen
make[3]: *** Es wurden keine Ziele angegeben und keine „make“-Steuerdatei gefunden.  Schluss.
CMake Error at /home/ecke/Downloads/VeloView/pv-veloview-build.cmake:26 (message):
  Failed!!!

CMakeFiles/veloview.dir/build.make:116: die Regel für Ziel „veloview/src/veloview-stamp/veloview-build“ scheiterte
make[2]: *** [veloview/src/veloview-stamp/veloview-build] Fehler 1
make[2]: Verzeichnis „/home/ecke/Downloads/VeloView“ wird verlassen
CMakeFiles/Makefile2:141: die Regel für Ziel „CMakeFiles/veloview.dir/all“ scheiterte
make[1]: *** [CMakeFiles/veloview.dir/all] Fehler 2
make[1]: Verzeichnis „/home/ecke/Downloads/VeloView“ wird verlassen
Makefile:149: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2

bastienjacquet, 2017-11-09:
Hello,
The code corresponding to VeloView 3.5.0 binaries has been released.
It has been tested on Ubuntu 14.04, Ubuntu 16.04, Windows 10 (32/64), MacOS.

Please open a new issue if you encounter any problems.

Best regards,
Bastien Jacquet
VeloView Lead Developer

closed 2017-11-09 by bastienjacquet