Kitware/VeloView #25: Paraview clone issues

GirijaByraiah, 2016-12-27:
Trying to build veloview as per instructions on windows 7 machine with visual studio 9 and latest CMAKE 3.7.1 ,but i am stuck with the following errors. Please help me fix this!
Performing download step (git clone) for 'paraview' 2>Cloning into 'paraview'... 2>Checking out files: 12% (917/7508) Checking out files: 13% (977/7508) Checking out files: 14% (1052/7508) Checking out files: 15% (1127/7508) Checking out files: 16% (1202/7508) Checking out files: 17% (1277/7508) Checking out files: 18% (1352/7508) Checking out files: 19% (1427/7508) Checking out files: 20% (1502/7508) Checking out files: 21% (1577/7508) Checking out files: 22% (1652/7508) Checking out files: 23% (1727/7508) Checking out files: 24% (1802/7508) Checking out files: 25% (1877/7508) Checking out files: 26% (1953/7508) Checking out files: 27% (2028/7508) Checking out files: 28% (2103/7508) Checking out files: 29% (2178/7508) Checking out files: 30% (2253/7508) Checking out files: 30% (2316/7508) Checking out files: 31% (2328/7508) Checking out files: 32% (2403/7508) Checking out files: 33% (2478/7508) Checking out files: 34% (2553/7508) Checking out files: 35% (2628/7508) Checking out files: 36% (2703/7508) Checking out files: 37% (2778/7508) Checking out files: 38% (2854/7508) Checking out files: 39% (2929/7508) Checking out files: 40% (3004/7508) Checking out files: 41% (3079/7508) Checking out files: 42% (3154/7508) Checking out files: 43% (3229/7508) Checking out files: 44% (3304/7508) Checking out files: 45% (3379/7508) Checking out files: 45% (3433/7508) Checking out files: 46% (3454/7508) Checking out files: 47% (3529/7508) Checking out files: 47% (3599/7508) Checking out files: 48% (3604/7508) Checking out files: 49% (3679/7508) Checking out files: 50% (3754/7508) Checking out files: 51% (3830/7508) Checking out files: 52% (3905/7508) Checking out files: 53% (3980/7508) Checking out files: 54% (4055/7508) Checking out files: 55% (4130/7508) Checking out files: 56% (4205/7508) Checking out files: 57% (4280/7508) Checking out files: 58% (4355/7508) Checking out files: 59% (4430/7508) Checking out files: 60% (4505/7508) Checking out files: 61% (4580/7508) Checking out files: 62% (4655/7508) Checking out files: 63% (4731/7508) Checking out files: 64% (4806/7508) Checking out files: 64% (4829/7508) Checking out files: 65% (4881/7508) Checking out files: 66% (4956/7508) Checking out files: 67% (5031/7508) Checking out files: 68% (5106/7508) Checking out files: 69% (5181/7508) Checking out files: 70% (5256/7508) Checking out files: 71% (5331/7508) Checking out files: 72% (5406/7508) Checking out files: 73% (5481/7508) Checking out files: 74% (5556/7508) Checking out files: 75% (5631/7508) Checking out files: 76% (5707/7508) Checking out files: 77% (5782/7508) Checking out files: 78% (5857/7508) Checking out files: 79% (5932/7508) Checking out files: 80% (6007/7508) Checking out files: 81% (6082/7508) Checking out files: 82% (6157/7508) Checking out files: 83% (6232/7508) Checking out files: 84% (6307/7508) Checking out files: 85% (6382/7508) Checking out files: 86% (6457/7508) Checking out files: 87% (6532/7508) Checking out files: 87% (6543/7508) Checking out files: 88% (6608/7508) Checking out files: 89% (6683/7508) Checking out files: 90% (6758/7508) Checking out files: 91% (6833/7508) Checking out files: 92% (6908/7508) Checking out files: 93% (6983/7508) Checking out files: 94% (7058/7508) Checking out files: 95% (7133/7508) Checking out files: 96% (7208/7508) Checking out files: 97% (7283/7508) Checking out files: 98% (7358/7508) Checking out files: 99% (7433/7508) Checking out files: 100% (7508/7508) Checking out files: 100% (7508/7508), done. 2>Note: checking out 'origin/veloview'. 2>You are in 'detached HEAD' state. You can look around, make experimental 2>changes and commit them, and you can discard any commits you make in this 2>state without impacting any branches by performing another checkout. 2>If you want to create a new branch to retain commits you create, you may 2>do so (now or later) by using -b with the checkout command again. Example: 2> git checkout -b 2>HEAD is now at f60010ca... Merge commit 'c1c6fbb8' into fixedrep 2>Submodule 'ThirdParty/IceT/vtkicet' (git://paraview.org/IceT.git) registered for path 'ThirdParty/IceT/vtkicet' 2>Submodule 'ThirdParty/QtTesting/vtkqttesting' (git://paraview.org/QtTesting.git) registered for path 'ThirdParty/QtTesting/vtkqttesting' 2>Submodule 'ThirdParty/protobuf/vtkprotobuf' (git://public.kitware.com/protobuf.git) registered for path 'ThirdParty/protobuf/vtkprotobuf' 2>Submodule 'Utilities/VisItBridge' (git://paraview.org/VisItBridge.git) registered for path 'Utilities/VisItBridge' 2>Submodule 'VTK' (git://vtk.org/VTK.git) registered for path 'VTK' 2>Cloning into 'C:/Users/pjcy0j/VeloView/build/paraview/src/paraview/ThirdParty/IceT/vtkicet'... 2>fatal: unable to access 'https://paraview.org/IceT.git/': SSL: no alternative certificate subject name matches target host name 'paraview.org' 2>fatal: clone of 'git://paraview.org/IceT.git' into submodule path 'C:/Users/pjcy0j/VeloView/build/paraview/src/paraview/ThirdParty/IceT/vtkicet' failed 2>Failed to clone 'ThirdParty/IceT/vtkicet'. Retry scheduled 2>Cloning into 'C:/Users/pjcy0j/VeloView/build/paraview/src/paraview/ThirdParty/QtTesting/vtkqttesting'... 2>fatal: unable to access 'https://paraview.org/QtTesting.git/': SSL: no alternative certificate subject name matches target host name 'paraview.org' 2>fatal: clone of 'git://paraview.org/QtTesting.git' into submodule path 'C:/Users/pjcy0j/VeloView/build/paraview/src/paraview/ThirdParty/QtTesting/vtkqttesting' failed 2>Failed to clone 'ThirdParty/QtTesting/vtkqttesting'. Retry scheduled 2>Cloning into 'C:/Users/pjcy0j/VeloView/build/paraview/src/paraview/ThirdParty/protobuf/vtkprotobuf'... 2>fatal: repository 'https://public.kitware.com/protobuf.git/' not found 2>fatal: clone of 'git://public.kitware.com/protobuf.git' into submodule path 'C:/Users/pjcy0j/VeloView/build/paraview/src/paraview/ThirdParty/protobuf/vtkprotobuf' failed 2>Failed to clone 'ThirdParty/protobuf/vtkprotobuf'. Retry scheduled 2>Cloning into 'C:/Users/pjcy0j/VeloView/build/paraview/src/paraview/Utilities/VisItBridge'... 2>fatal: unable to access 'https://paraview.org/VisItBridge.git/': SSL: no alternative certificate subject name matches target host name 'paraview.org' 2>fatal: clone of 'git://paraview.org/VisItBridge.git' into submodule path 'C:/Users/pjcy0j/VeloView/build/paraview/src/paraview/Utilities/VisItBridge' failed 2>Failed to clone 'Utilities/VisItBridge'. Retry scheduled 2>Cloning into 'C:/Users/pjcy0j/VeloView/build/paraview/src/paraview/VTK'... 2>fatal: unable to access 'https://vtk.org/VTK.git/': SSL: no alternative certificate subject name matches target host name 'vtk.org' 2>fatal: clone of 'git://vtk.org/VTK.git' into submodule path 'C:/Users/pjcy0j/VeloView/build/paraview/src/paraview/VTK' failed 2>Failed to clone 'VTK'. Retry scheduled 2>Cloning into 'C:/Users/pjcy0j/VeloView/build/paraview/src/paraview/ThirdParty/IceT/vtkicet'... 2>fatal: unable to access 'https://paraview.org/IceT.git/': SSL: no alternative certificate subject name matches target host name 'paraview.org' 2>fatal: clone of 'git://paraview.org/IceT.git' into submodule path 'C:/Users/pjcy0j/VeloView/build/paraview/src/paraview/ThirdParty/IceT/vtkicet' failed 2>Failed to clone 'ThirdParty/IceT/vtkicet' a second time, aborting 2>CMake Error at C:/Users/pjcy0j/VeloView/build/paraview/tmp/paraview-gitclone.cmake:83 (message): 2> Failed to update submodules in: 2> 'C:/Users/pjcy0j/VeloView/build/paraview/src/paraview' 2>Project : error PRJ0019: A tool returned an error code from "Performing download step (git clone) for 'paraview'" 2>Build log was saved at "file://c:\Users\pjcy0j\VeloView\build\paraview.dir\Debug\BuildLog.htm" 2>paraview - 1 error(s), 0 warning(s) ========== Build: 1 succeeded, 1 failed, 3 up-to-date, 0 skipped ==========

bastienjacquet, 2016-12-30:
Hello Girija,
VeloView 3.1 is based on Paraview 4.2. The 4.2 ParaView version is unmaintained now, and its submodules are failing to download.
We do have an updated version internally, but it's waiting upon release approval.
I will let you know about any progress asap.
Best,
Bastien,
VeloView Lead Developer

bastienjacquet, 2017-11-09:
Hello,
The code corresponding to VeloView 3.5.0 binaries has been released.
Best regards,
Bastien Jacquet
VeloView Lead Developer

closed 2017-11-09 by bastienjacquet