Kitware/VeloView #26: Error starting VeloView

Grillen80, 2017-02-07:
I've build the VeloView Application without any errors under Linux Xenial 16.04.
While starting the application, the following error occurs and I do not have any possibilties to choose a device. 

_Traceback (most recent call last):
  File "<console>", line 1, in <module>
ImportError: No module named paraview.simple
Traceback (most recent call last):
  File "<string>", line 4, in <module>
  File "/usr/local/lib/veloview-3.1/site-packages/veloview/applogic.py", line 20, in <module>
    import paraview.simple as smp
ImportError: No module named paraview.simple
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'vv' is not defined_

Thanks for your help in advance. Kind regards,
Sven

Grillen80, 2017-02-09:
I fixed the issue with an export of the installed site-packages to the $PYTHONPATH

closed 2017-02-09 by Grillen80

karnikram, 2017-08-28:
Hi, 
I'm facing the same issue.

Could you provide me more details about your $PYTHONPATH?