Kitware/VeloView #27: Linux Veloview configuration file VLP 16

Grillen80, 2017-02-09:
Finally the Veloview application starts without any errors under Linux 16.04. When I choose a device I get the following error:

> ERROR: In .../VeloView/VelodyneHDL/vtkVelodyneHDLReader.cxx, line 523
vtkVelodyneHDLReader (0x670e770): Invalid sensor configuration file/usr/local/lib/veloview-3.1/../share/VLP-16.xml_

There is no existing "share" folder with a VLP-16.xml file. Did I missed some configurations in the makefile or do I have to download them from a different source? 

Thanks for your help,
Sven

lurenlym, 2017-09-11:
did you solve this problem now?i meet the same issue.

lupusorina, 2017-09-19:
I also have this problem. Any solution?

bastienjacquet, 2017-11-08:
Hello,
The code corresponding to VeloView 3.5.0 binaries has been released here.
This issue should not happen anymore, so please re-open a new one if necessary.
Best regards,
Bastien Jacquet
VeloView Lead Developer

closed 2017-11-08 by bastienjacquet