Kitware/VeloView #28: Veloview fails to execute on Ubuntu 16.04

Chambana, 2017-02-24:
fairly new install of 16.04.2 LTS and simply can't get VeloView to work.  Here's what I've done:

**clone the source:**
git clone https://github.com/Kitware/VeloView.git

**make build directory:**
cd VeloView
mkdir build

**Configure the project:**
~/VeloView/Superbuild$ cmake ../../VeloView/Superbuild/ -DPYTHON_EXECUTABLE:FILEPATH=/usr/bin/python2.7 -DPYTHON_LIBRARY:FILEPATH=/usr/lib/python2.7/config-x86_64-linux-gnu/libpython2.7.so -DPYTHON_INCLUDE_DIR:FILEPATH=/usr/include/python2.7/ -DENABLE_veloview=ON -DUSE_SYSTEM_boost=ON -DUSE_SYSTEM_python=ON -DUSE_SYSTEM_qt=ON

-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check size of void*
-- Check size of void* - done
-- Enabling qt since needed by: paraview;pythonqt;veloview
-- Enabling python since needed by: paraview;pythonqt
-- Enabling pythonqt since needed by: paraview
-- Enabling paraview since needed by: veloview
-- Enabling pcap since needed by: veloview
-- Enabling boost since needed by: liblas;veloview
-- Enabling eigen since needed by: veloview
-- Enabling liblas since needed by: veloview
-- Enabling veloview as requested.
-- PROJECTS_ENABLED qt;python;pythonqt;paraview;pcap;boost;eigen;liblas;veloview
-- Found PythonLibs: /usr/lib/python2.7/config-x86_64-linux-gnu/libpython2.7.so (found version "2.7.12") 
-- Found PythonInterp: /usr/bin/python2.7 (found version "2.7.12") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/username/VeloView/Superbuild

**Build the Project:**
make -j4

.....
[ 98%] Performing install step for 'veloview'
[  6%] Built target python_byte_compile
[ 51%] Built target VelodyneHDLPlugin
[ 52%] Built target PacketFileSender
[ 54%] Built target TestPositionReader
[ 58%] Built target TestSensor
[ 73%] Built target VelodyneHDLPluginPythonD
[ 75%] Built target TestReader
[ 81%] Built target VelodyneHDLPythonQT
[ 83%] Built target VelodyneHDLPluginPython
[ 99%] Built target VeloView
[100%] Built target VeloView-launcher
Install the project...
-- Install configuration: ""
-- Installing: /home/username/VeloView/Superbuild/install/share/VLP-16.xml
-- Installing: /home/username/VeloView/Superbuild/install/share/HDL-32.xml
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/VeloView
-- Set runtime path of "/home/username/VeloView/Superbuild/install/lib/veloview-3.1/VeloView" to "/usr/lib/python2.7/config-x86_64-linux-gnu"
-- Installing: /home/username/VeloView/Superbuild/install/bin/VeloView
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/libVelodyneHDLPlugin.so
-- Set runtime path of "/home/username/VeloView/Superbuild/install/lib/veloview-3.1/libVelodyneHDLPlugin.so" to "/usr/lib/python2.7/config-x86_64-linux-gnu"
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/VelodyneHDLPluginPython.so
-- Set runtime path of "/home/username/VeloView/Superbuild/install/lib/veloview-3.1/VelodyneHDLPluginPython.so" to "/usr/lib/python2.7/config-x86_64-linux-gnu"
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/libVelodyneHDLPluginPythonD.so
-- Set runtime path of "/home/username/VeloView/Superbuild/install/lib/veloview-3.1/libVelodyneHDLPluginPythonD.so" to "/usr/lib/python2.7/config-x86_64-linux-gnu"
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/libVelodyneHDLPythonQT.so
-- Set runtime path of "/home/username/VeloView/Superbuild/install/lib/veloview-3.1/libVelodyneHDLPythonQT.so" to "/usr/lib/python2.7/config-x86_64-linux-gnu"
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/site-packages/veloview
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/site-packages/veloview/timer.py
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/site-packages/veloview/__init__.py
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/site-packages/veloview/gridAdjustmentDialog.py
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/site-packages/veloview/timer.pyc
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/site-packages/veloview/__init__.pyc
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/site-packages/veloview/planefit.pyc
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/site-packages/veloview/applogic.pyc
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/site-packages/veloview/planefit.py
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/site-packages/veloview/kiwiviewerExporter.pyc
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/site-packages/veloview/kiwiviewerExporter.py
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/site-packages/veloview/applogic.py
-- Installing: /home/username/VeloView/Superbuild/install/lib/veloview-3.1/site-packages/veloview/gridAdjustmentDialog.pyc
[100%] Completed 'veloview'
[100%] Built target veloview

**Attempt to execute Veloview:**
~/VeloView/Superbuild/install/bin$ ./VeloView 

/home/username/VeloView/Superbuild/install/lib/veloview-3.1/VeloView: error while loading shared libraries: libPythonQt.so: cannot open shared object file: No such file or directory

**I then tried:**
~/VeloView/Superbuild$ sudo make install

...
-- Installing: /usr/local/lib/veloview-3.1/site-packages/veloview/applogic.py
-- Installing: /usr/local/lib/veloview-3.1/site-packages/veloview/gridAdjustmentDialog.pyc
-- Installing: /usr/local/bin/VeloView
[it appears to install successfully]

When I try to execute VeloView from /usr/loca/bin/Veloview, I get that same error:

~/VeloView/Superbuild$ /usr/local/bin/VeloView 
/usr/local/lib/veloview-3.1/VeloView: error while loading shared libraries: libPythonQt.so: cannot open shared object file: No such file or directory

Any ideas?  The project seemed build successfully, but I can't execute it.  I can launch all the other programs in the install directory (e.g. paraview), but not Veloview.  Of note, the file libPythonQt.so is present in 2 directories:

~/Veloview/Superbuild/install/lib
and
~/Velowview/Superbuild/pythonqt/src/pythonqt-build

nightsparc, 2017-06-06:
The missing libPythonQt.so is located in the build/install/lib dir.
Just point the runtime linker to it (e.g. by setting LD_LIBRARY_PATH), which will allow you to run VeloView

saulogrego, 2017-06-15:
@nightsparc, how can i do this (point the runtime linker)? I'm with the same issue and i'm new on Linux/VeloView. Thanks in advance.

nightsparc, 2017-06-15:
@saulogrego 

It's quite easy :) on the terminal enter:
```bash

export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:$PWD/../lib:$PWD/../lib/pandar-1.0:$PWD/../lib/paraview-4.3
```

Or you use a script. To start it. @trickyMan provided a script in: https://github.com/Kitware/VeloView/issues/29

which i slighly modified for my needs and put in the `VeloView/build/install/bin` folder

```bash
#!/bin/sh
# From: https://github.com/Kitware/VeloView/issues/29

LP=$PWD/../lib

export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:${LP}:${LP}/pandar-1.0:${LP}/paraview-4.3
export PYTHONPATH=${PYTHONPATH}:${LP}/paraview-4.3/site-packages:${LP}/pandar-1.0/site-packages:${LP}/paraview-4.3/site-packages/vtk/
./VeloView
```

saulogrego, 2017-06-15:
Thanks @nightsparc. VeloView its working now.

bastienjacquet, 2017-11-09:
Hello,
The code corresponding to VeloView 3.5.0 binaries has been released.
This workaround should not be needed anymore.
Closing the issue for now.
Best regards,
Bastien Jacquet
VeloView Lead Developer

closed 2017-11-09 by bastienjacquet