Kitware/VeloView #32: can‘t make on ubuntu

lurenlym, 2017-09-09:
my system is ubuntu 17.04 and when i make the project it seem the paraview can't build well

[ 40%] Completed 'pythonqt'
[ 44%] Built target pythonqt
[ 45%] Performing update step for 'paraview'
Current branch HEAD is up to date.
[ 47%] Performing configure step for 'paraview'
-- Setting build type to 'Debug' as none was specified.
-- Determined Source Version : 4.3.1-63-gf60010ca65
CMake Error at VTK/CMake/vtkCompilerExtras.cmake:42 (if):
  if given arguments:

    "cc: error: ARGS: No such file or directory

  cc (Ubuntu 6.3.0-12ubuntu2) 6.3.0 20170406

  Copyright (C) 2016 Free Software Foundation, Inc.

  This is free software" " see the source for copying conditions.  There is
  NO

  warranty" " not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR
  PURPOSE.

  

  " "VERSION_GREATER" "4.2.0" "AND" "BUILD_SHARED_LIBS" "AND"
  "HAVE_GCC_VISIBILITY" "AND" "VTK_USE_GCC_VISIBILITY" "AND" "NOT" "MINGW"
  "AND" "NOT" "CYGWIN"

  Unknown arguments specified
Call Stack (most recent call first):
  VTK/CMakeLists.txt:244 (include)

-- Configuring incomplete, errors occurred!
See also "/home/lyming/Myproject/build/paraview/src/paraview-build/CMakeFiles/CMakeOutput.log".
See also "/home/lyming/Myproject/build/paraview/src/paraview-build/CMakeFiles/CMakeError.log".
CMakeFiles/paraview.dir/build.make:108: recipe for target 'paraview/src/paraview-stamp/paraview-configure' failed
make[2]: *** [paraview/src/paraview-stamp/paraview-configure] Error 1
CMakeFiles/Makefile2:106: recipe for target 'CMakeFiles/paraview.dir/all' failed
make[1]: *** [CMakeFiles/paraview.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

lurenlym, 2017-09-11:
i build it now.May be it is some tips  to help someone solve this problem.I found the paraview packages isn't well configure by gnu 6.4 And ubuntu 17.04 used this version.i choose the ubuntu 16.04 which used gnu 5.4 and it build paraview well

closed 2017-09-11 by lurenlym

Francisobiagwu, 2019-01-08:
I currently have the same exact issue.