Kitware/VeloView #33: veloview 3.5 source code?

moskewcz, 2017-09-26:
binaries of veloview 3.5 have been released for a little while now here:

https://www.paraview.org/Wiki/VeloView

but, it seems like the source code is nowhere to be found. anyone have any info/hints/advice?

bastienjacquet, 2017-10-09:
Hello,
The code corresponding to VeloView 3.5 binaries will soon be released.
It just need some cleanup of the metadata for the commits not to interfere with github issues and merge requests.
Best regards,
Bastien Jacquet
VeloView Lead Developer

moskewcz, 2017-10-10:
thanks for the update! i guess i'll close this if/when the 3.5 source show up ... ;)

bastienjacquet, 2017-11-08:
Hello,
The code corresponding to VeloView 3.5.0 binaries has been released.
Best regards,
Bastien Jacquet
VeloView Lead Developer

closed 2017-11-08 by bastienjacquet

KrantiYadav, 2018-11-20:
Hi all,

I am new here for Veloview. I am looking for solution in python which will convert .pcap to .las.

Any help here?

Regards,
Kranti