Kitware/VeloView #35: Cloning into 'paraview' permission denied

mpkuse, 2017-11-14:
I am trying to view my lidar data with VeloView unable to succeed yet. I have a Velodyne VLP-16. My workstation is Ubuntu 16.04.

I tried to clone this repo and SuperBuild it following the doc.

```cmake -DENABLE_veloview:BOOL=ON -DUSE_SYSTEM_python=ON  ../VeloView/Superbuild/```

``` make ```

I get the following error:

```
[ 68%] Performing download step (git clone) for 'paraview'
Cloning into 'paraview'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
Cloning into 'paraview'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
Cloning into 'paraview'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
-- Had to git clone more than once:
          3 times.
CMake Error at /home/mpkuse/Downloads/build/paraview/tmp/paraview-gitclone.cmake:40 (message):
  Failed to clone repository: 'git@gitlab.kitware.com:bjacquet/paraview.git'

CMakeFiles/paraview.dir/build.make:89: recipe for target 'paraview/src/paraview-stamp/paraview-download' failed
make[2]: *** [paraview/src/paraview-stamp/paraview-download] Error 1
CMakeFiles/Makefile2:479: recipe for target 'CMakeFiles/paraview.dir/all' failed
make[1]: *** [CMakeFiles/paraview.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

bastienjacquet, 2017-11-14:
Hello,

I'm not sure why, but you got a "Permission denied (publickey)." from the server, while downloading the Paraview subproject.
The repository is public though : https://gitlab.kitware.com/bjacquet/paraview
I'll check this.
Best regards,
Bastien Jacquet - VeloView Lead Developer

mpkuse, 2017-11-14:
do you think I should cmake again?

mpkuse, 2017-11-14:
I tried to deleted the ~/.ssh/know_hosts file and retry, still didn't work.

Although I can confirm that I can clone this repo with git clone with http-access
in a standalone way. Any way to get around this issue?

```
[ 68%] Performing download step (git clone) for 'paraview'
Cloning into 'paraview'...
The authenticity of host 'gitlab.kitware.com (66.162.65.214)' can't be established.
ECDSA key fingerprint is SHA256:Lm8wR6C3yccDC25Og9xIpS+WtfKJdB1pVtAhdN82v0Q.
Are you sure you want to continue connecting (yes/no)? y
Please type 'yes' or 'no': yes
Warning: Permanently added 'gitlab.kitware.com,66.162.65.214' (ECDSA) to the list of known hosts.
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
Cloning into 'paraview'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
Cloning into 'paraview'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
-- Had to git clone more than once:
          3 times.
CMake Error at /home/mpkuse/Downloads/build/paraview/tmp/paraview-gitclone.cmake:40 (message):
  Failed to clone repository: 'git@gitlab.kitware.com:bjacquet/paraview.git'

CMakeFiles/paraview.dir/build.make:89: recipe for target 'paraview/src/paraview-stamp/paraview-download' failed
make[2]: *** [paraview/src/paraview-stamp/paraview-download] Error 1
CMakeFiles/Makefile2:479: recipe for target 'CMakeFiles/paraview.dir/all' failed
make[1]: *** [CMakeFiles/paraview.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

bastienjacquet, 2017-11-14:
Creating an account on our open servers gitlab.kitware.com and putting your
ssh pubkey would solve this.
As a workaround, you can also change the url in Superbuild/versions.cmake
from "git@gitlab.kitware.com:bjacquet/paraview.git" to
"*https://gitlab.kitware.com/bjacquet/paraview.git
<https://gitlab.kitware.com/bjacquet/paraview.git>*"

>

mpkuse, 2017-11-14:
After adding my ssh key into gitlab kitware servers I am able to download the paraview repository etc. 

It seem to compile well. However I encounter another error. I believe in paraview. Something to do with python2 and python3 seem like. 

Any suggestion please.

```
 81%] Performing build step for 'paraview'
[  0%] Built target ParaViewData
[  0%] Built target vtkWrappingTools
[  0%] Built target vtksys
[  0%] Compiling Python package 'pygments'
[  0%] Built target VTKData
[  0%] Built target vtkalglib
[  0%] Built target vtkDICOMParser
[  1%] Built target vtkzlib
Scanning dependencies of target vtkEncodeString
Scanning dependencies of target vtkParseOGLExt
[  1%] Built target vtkjpeg
[  1%] Built target vtkmkg3states
[  1%] Building CXX object VTK/Utilities/ParseOGLExt/CMakeFiles/vtkParseOGLExt.dir/Tokenizer.cxx.o
[  1%] Building CXX object VTK/Utilities/ParseOGLExt/CMakeFiles/vtkParseOGLExt.dir/ParseOGLExt.cxx.o
[  1%] Building CXX object VTK/Utilities/EncodeString/CMakeFiles/vtkEncodeString.dir/vtkEncodeString.cxx.o
Scanning dependencies of target vtkHashSource
[  1%] Building CXX object VTK/Utilities/HashSource/CMakeFiles/vtkHashSource.dir/vtkHashSource.cxx.o
Listing '/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments'...
Compiling '/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/__init__.py'...
***   File "/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/__init__.py", line 46
    except TypeError, err:
                    ^
SyntaxError: invalid syntax

Compiling '/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/cmdline.py'...
***   File "/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/cmdline.py", line 122
    print "Help on the %s lexer:" % cls.name
                                ^
SyntaxError: Missing parentheses in call to 'print'

Listing '/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/filters'...
Listing '/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters'...
Compiling '/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters/_mapping.py'...
***   File "/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters/_mapping.py", line 46
    print module_name
                    ^
SyntaxError: Missing parentheses in call to 'print'

Compiling '/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters/html.py'...
***   File "/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters/html.py", line 518
    except IOError, err:
                  ^
SyntaxError: invalid syntax

Compiling '/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/lexer.py'...
***   File "/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/lexer.py", line 456
    except Exception, err:
                    ^
SyntaxError: invalid syntax

Listing '/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/lexers'...
Compiling '/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/lexers/_mapping.py'...
***   File "/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/lexers/_mapping.py", line 36
    print module_name
                    ^
SyntaxError: Missing parentheses in call to 'print'

Compiling '/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/lexers/agile.py'...
***   File "/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/lexers/agile.py", line 340
    re.match(ur'  File "[^"]+", line \d+\n$', line)):
                                           ^
SyntaxError: invalid syntax

Listing '/home/mpkuse/Downloads/build/paraview/src/paraview-build/lib/site-packages/pygments/styles'...
ThirdParty/pygments/CMakeFiles/Pygments.dir/build.make:60: recipe for target 'ThirdParty/pygments/pygments.build-complete' failed
make[5]: *** [ThirdParty/pygments/pygments.build-complete] Error 1
CMakeFiles/Makefile2:853: recipe for target 'ThirdParty/pygments/CMakeFiles/Pygments.dir/all' failed
make[4]: *** [ThirdParty/pygments/CMakeFiles/Pygments.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
[  1%] Linking CXX executable ../../../bin/vtkHashSource-pv5.1
[  1%] Built target vtkHashSource
[  1%] Linking CXX executable ../../../bin/vtkEncodeString-pv5.1
[  1%] Built target vtkEncodeString
[  1%] Linking CXX executable ../../../bin/vtkParseOGLExt-pv5.1
[  1%] Built target vtkParseOGLExt
Makefile:127: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMake Error at /home/mpkuse/Downloads/build/pv-paraview-build.cmake:26 (message):
  Failed!!!

CMakeFiles/paraview.dir/build.make:113: recipe for target 'paraview/src/paraview-stamp/paraview-build' failed
make[2]: *** [paraview/src/paraview-stamp/paraview-build] Error 1
CMakeFiles/Makefile2:479: recipe for target 'CMakeFiles/paraview.dir/all' failed
make[1]: *** [CMakeFiles/paraview.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

tsyip88, 2017-11-15:
I also ran into the issue of failing to clone the repository using ssh. Something must have changed recently since I was able to clone the repo using the same command around a month ago. Updating the link in the versions.cmake to use https as suggested seemed to fix it.

bastienjacquet, 2017-11-16:
Hello, 
We now have updated master to switch to https. See c52ba9024e2f625d6edba67fb473523f70aab22f.
This closes the issue.
Best regards,
Bastien Jacquet,
VeloView Lead Developer

closed 2017-11-16 by bastienjacquet