Kitware/VeloView #36: Cmake version specified in installation instructions cannot build package

tsyip88, 2017-11-15:
The doc indicates that cmake needs to be installed using the following steps: 
```sudo apt-get install software-properties-common
sudo add-apt-repository ppa:george-edison55/cmake-3.x
sudo apt-get update
sudo apt-get install --reinstall cmake
```

Unfortunately, this no longer works since a recent change pushed the required version of cmake to at least 3.3 while the cmake version in the ppa is 3.2.2.2~ubuntu14.04.1

Maybe docs be updated to run `sudo apt-get install cmake3` instead?

Chatmetaleux, 2018-03-23:
FYI, 
I had to change the `cmake_minimum_required(VERSION 3.2)`
in the file _CMakeLists.txt_ in _build/paraview/src/paraview_ folder to be able to compile paraview...
Also followed the tuto with Ubuntu 14.04.1.

vnn0804, 2019-02-17:
hey hi try 16.04 ubuntu. and build dir not required. follow the following steps

1. First, install the following dependencies using the apt-get commands
   git
   cmake-curses-gui
   build-essential
   libboost-all-dev
   libxt-dev
   libbz2-dev
   libqt4-dev
   qt4-default
   qt4-dev-tools
   zlib1g-dev
   libpcap-dev.
   export QT_SELECT=qt4
2. create project dir
   mkdir project
3. clone the veloview project
   git clone https://github.com/Kitware/VeloView.git
4. cmake build type following commands
   a) ccmake VeloView/Superbuild
   b) type c to configure
   c) page 1 for cmake
   CMAKE_BUILD_TYPE                *Release
   CMAKE_INSTALL_PREFIX            */home/icarus/project/
   ENABLE_boost                    *OFF
   ENABLE_eigen                    *OFF
   ENABLE_liblas                   *OFF
   ENABLE_paraview                 *OFF
   ENABLE_pcap                     *OFF
   ENABLE_python                   *OFF
   ENABLE_pythonqt                 *OFF
   ENABLE_qt                       *OFF
   ENABLE_veloview                 *ON
   qt_EXTRA_CONFIGURATION_OPTIONS  *

again type c (configure) and type e for exit

d) page 2 for cmake
USE_SYSTEM_boost *ON
USE_SYSTEM_pcap *ON
USE_SYSTEM_python *ON
USE_SYSTEM_qt *ON
CMAKE_BUILD_TYPE Release
CMAKE_INSTALL_PREFIX /home/icarus/project
ENABLE_veloview ON
qt_EXTRA_CONFIGURATION_OPTIONS

type c again type c and finally type g

1. run  cmake -DENABLE_veloview:BOOL=ON VeloView/Superbuild
2. run  make -j4 and make in two different terminal
3. finally open veloview application--  cd install/bin
   ./VeloView