Kitware/VeloView #37: missing include files

mocyth, 2017-11-24:
Recently, I got problems when trying to build paraview project, whick are described as follows:

    218>  pqLineEditEventPlayer.cxx
    218>e:\0_open_sources\kitware\paraview\qt\widgets\pqLineEditEventPlayer.h(35): fatal error C1083: can't open include file: “pqAbstractStringEventPlayer.h”: No such file or directory
    218>  pqListWidgetCheckHelper.cxx
    218>  pqTreeWidget.cxx
    218>E:\0_open_sources\Kitware\ParaView\Qt\Widgets\pqTreeWidget.cxx(45): fatal error C1083: can't open include file: “pqTimer.h”: No such file or directory
    218>  pqTreeWidgetItem.cxx
    218>  moc_pqColorButtonEventPlayer.cpp
    218>e:\0_open_sources\kitware\paraview\build\qt\widgets\../../../Qt/Widgets/pqColorButtonEventPlayer.h(35): fatal error C1083: can't open include file: “pqWidgetEventPlayer.h”: No such file or directory
    218>  moc_pqColorButtonEventTranslator.cpp
    218>e:\0_open_sources\kitware\paraview\build\qt\widgets\../../../Qt/Widgets/pqColorButtonEventTranslator.h(35): fatal error C1083: can't open include file: “pqWidgetEventTranslator.h”: No such file or directory
    218>  moc_pqColorChooserButton.cpp
    218>  moc_pqColorDialogEventPlayer.cpp
    218>e:\0_open_sources\kitware\paraview\build\qt\widgets\../../../Qt/Widgets/pqColorDialogEventPlayer.h(35): fatal error C1083: can't open include file: “pqWidgetEventPlayer.h”: No such file or directory
    218>  moc_pqColorDialogEventTranslator.cpp
    218>e:\0_open_sources\kitware\paraview\build\qt\widgets\../../../Qt/Widgets/pqColorDialogEventTranslator.h(35): fatal error C1083: can't open include file: “pqWidgetEventTranslator.h”: No such file or directory
    218>  moc_pqConsoleWidgetEventPlayer.cpp
    218>e:\0_open_sources\kitware\paraview\build\qt\widgets\../../../Qt/Widgets/pqConsoleWidgetEventPlayer.h(35): fatal error C1083: can't open include file: “pqWidgetEventPlayer.h”: No such file or directory

After checking the source repository, i didn't find these include files either.
Would you tell me where to git-clone or download these include files.
Thanks very very much and appreciate your excellent work.

bastienjacquet, 2017-11-24:
Hello,
It looks like some Qt files are missing. Are you using a system-wide install or building it as a VeloView sub-project ?
If you have a system install, then you should check that the version you are using contains all needed files.
If you are using the version automatically build from the VeloView sources, then ... you probably found a bug. Please report your full configuration so we can investigate.

Best regards,
Bastien Jacquet - Lead VeloView Developer

mocyth, 2017-11-24:
I'm really sorry that i didnot write correctly down my project environment. Actually, i was previously building up paraview sdk, then i can build up veloview project. However, i got the problems as described before, when i was building paraview, and i found there indeed are no those .h files in paraview repository.

Really apreciate your attention and replay , and happy thanksgiving day.