Kitware/VeloView #39: Superbuild on Windows 10 - veloview build failed

Anarchon-github, 2017-12-21:
Hi.
I've encountered a problem while building veloview -
it seems there is incorrect version of velodyheHDLplugin, in console log it says things like this:

`C:\Users\xrezv\Downloads\veloview\VelodyneHDL\vvPacketSender.cxx(113) : error C2039: 'gpsTimestamp' : is not a member of 'DataPacketFixedLength::HDLDataPacket'`
`c:\users\xrezv\downloads\veloview\velodynehdl\vtkDataPacket.h(109) : see declaration of 'DataPacketFixedLength::HDLDataPacket'`

I'm using Visual Studio 2008, with , Visual Studio 2008 Command Prompt (x86)
cmake v.3.10 configured with cmake-gui with following options:

ninja generator
CMAKE_BUILD_TYPE = Release
CMAKE_INSTALL_PREFIX = c:\vvbuildq
ENABLE_veloview = ON
USE_SYSTEM_qt = ON
QT_QMAKE_EXECUTABLE = C:/Qt/4.8.6/bin/qmake.exe
PACKAGE_SYSTEM_QT = ON

Reason I'm using system qt is because the superbuild version of qt
springs up with error when executing moc.exe during build

I've also tried building using x64 Cross Tools Command Prompt,
to no avail, 

Can you perhaps point me in right direction? I can elaborate further if need be

bastienjacquet, 2017-12-22:
Hello,

VeloView does compile using Visual Studio 2012. 
The problem you are encountering might be linked to the declaration of the member variable `gpsTimestamp` from `DataPacketFixedLength::HDLDataPacket` (line 113) as `uint32_t gpsTimestamp;`
Perhaps Visual Studio 2008 does not find the `uint32_t` type.

Can you check you don't have an earlier compilation error before this one?

Best regards,
Bastien Jacquet - VeloView Team

Anarchon-github, 2017-12-22:
Hello,
thanks for fast reply,

the first errors occur after 'building veloview' step:

```
[22/24] Performing build step for 'veloview'
FAILED: veloview/src/veloview-stamp/veloview-build 
cmd.exe /C "cd /D C:\vvbuildq\veloview\src\veloview-build && "C:\Users\xrezv\Downloads\cmake 3.1\bin\cmake.exe" --build . && "C:\Users\xrezv\Downloads\cmake 3.1\bin\cmake.exe" -E touch C:/vvbuildq/veloview/src/veloview-stamp/veloview-build"
[1/88] Generating vtkSMXML_VelodyneHDLPlugin.h

-- Generate module: ApplanixPositionReader

-- Generate module: ProcessingSample

-- Generate module: VelodyneHDLGridSource

-- Generate module: VelodyneHDLReader

-- Generate module: VelodyneHDLPositionReader

-- Generate module: VelodyneHDLSource

[2/88] Building CXX object VelodyneHDL\CMakeFiles\VelodyneHDLPlugin.dir\moc_pqVelodyneHDLSourceImplementation.cxx.obj

[3/88] Building CXX object VelodyneHDL\CMakeFiles\VelodyneHDLPlugin.dir\vtkRollingDataAccumulator.cxx.obj

FAILED: VelodyneHDL/CMakeFiles/VelodyneHDLPlugin.dir/vtkRollingDataAccumulator.cxx.obj 

C:\PROGRA~2\MICROS~1.0\VC\bin\cl.exe  /nologo /TP -DQT_CORE_LIB -DQT_DLL -DQT_GUI_LIB -DQT_NO_DEBUG -DVelodyneHDLPlugin_EXPORTS -Dinline=__inline -IC:\vvbuildq\paraview\src\paraview-build\Qt\ApplicationComponents -IC:\vvbuildq\paraview\src\paraview\Qt\ApplicationComponents -IC:\vvbuildq\paraview\src\paraview-build\Qt\Components -IC:\vvbuildq\paraview\src\paraview\Qt\Components -IC:\Qt\4.8.6\include\ActiveQt -IC:\Qt\4.8.6\include\QtDesigner -IC:\Qt\4.8.6\include\QtDeclarative -IC:\Qt\4.8.6\include\QtScriptTools -IC:\Qt\4.8.6\include\QtDBus -IC:\Qt\4.8.6\include\QtXml -IC:\Qt\4.8.6\include\QtSql -IC:\Qt\4.8.6\include\QtOpenGL -IC:\Qt\4.8.6\include\QtMultimedia -IC:\Qt\4.8.6\include\QtNetwork -IC:\Qt\4.8.6\include\phonon -IC:\Qt\4.8.6\include\QtXmlPatterns -IC:\Qt\4.8.6\include\QtWebKit -IC:\Qt\4.8.6\include\QtHelp -IC:\Qt\4.8.6\include\QtUiTools -IC:\Qt\4.8.6\include\QtTest -IC:\Qt\4.8.6\include\QtScript -IC:\Qt\4.8.6\include\QtSvg -IC:\Qt\4.8.6\include\Qt3Support -IC:\Qt\4.8.6\include\QtGui -IC:\Qt\4.8.6\include\QtCore -IC:\Qt\4.8.6\mkspecs\default -IC:\Qt\4.8.6\include -IC:\vvbuildq\paraview\src\paraview-build\Qt\Core -IC:\vvbuildq\paraview\src\paraview\Qt\Core -IC:\vvbuildq\paraview\src\paraview-build\Qt\Widgets -IC:\vvbuildq\paraview\src\paraview\Qt\Widgets -IC:\vvbuildq\paraview\src\paraview-build\VTK\GUISupport\Qt -IC:\vvbuildq\paraview\src\paraview\VTK\GUISupport\Qt -IC:\vvbuildq\paraview\src\paraview-build\VTK\Common\ExecutionModel -IC:\vvbuildq\paraview\src\paraview\VTK\Common\ExecutionModel -IC:\vvbuildq\paraview\src\paraview-build\VTK\Common\DataModel -IC:\vvbuildq\paraview\src\paraview\VTK\Common\DataModel -IC:\vvbuildq\paraview\src\paraview-build\VTK\Common\Math -IC:\vvbuildq\paraview\src\paraview\VTK\Common\Math -IC:\vvbuildq\paraview\src\paraview-build\VTK\Common\Core -IC:\vvbuildq\paraview\src\paraview\VTK\Common\Core -IC:\vvbuildq\paraview\src\paraview-build\VTK\Utilities\KWIML -IC:\vvbuildq\paraview\src\paraview\VTK\Utilities\KWIML -IC:\vvbuildq\paraview\src\paraview-build\VTK\Utilities\KWSys -IC:\vvbuildq\paraview\src\paraview\VTK\Utilities\KWSys -IC:\vvbuildq\paraview\src\paraview-build\VTK\Common\Misc -IC:\vvbuildq\paraview\src\paraview\VTK\Common\Misc -IC:\vvbuildq\paraview\src\paraview-build\VTK\Common\System -IC:\vvbuildq\paraview\src\paraview\VTK\Common\System -IC:\vvbuildq\paraview\src\paraview-build\VTK\Common\Transforms -IC:\vvbuildq\paraview\src\paraview\VTK\Common\Transforms -IC:\vvbuildq\paraview\src\paraview-build\VTK\Filters\Extraction -IC:\vvbuildq\paraview\src\paraview\VTK\Filters\Extraction -IC:\vvbuildq\paraview\src\paraview-build\VTK\Filters\Core -IC:\vvbuildq\paraview\src\paraview\VTK\Filters\Core -IC:\vvbuildq\paraview\src\paraview-build\VTK\Filters\General -IC:\vvbuildq\paraview\src\paraview\VTK\Filters\General -IC:\vvbuildq\paraview\src\paraview-build\VTK\Common\ComputationalGeometry -IC:\vvbuildq\paraview\src\paraview\VTK\Common\ComputationalGeometry -IC:\vvbuildq\paraview\src\paraview-build\VTK\Filters\Statistics -IC:\vvbuildq\paraview\src\paraview\VTK\Filters\Statistics -IC:\vvbuildq\paraview\src\paraview-build\VTK\Imaging\Fourier -IC:\vvbuildq\paraview\src\paraview\VTK\Imaging\Fourier -IC:\vvbuildq\paraview\src\paraview-build\VTK\Imaging\Core -IC:\vvbuildq\paraview\src\paraview\VTK\Imaging\Core -IC:\vvbuildq\paraview\src\paraview-build\VTK\ThirdParty\alglib -IC:\vvbuildq\paraview\src\paraview\VTK\ThirdParty\alglib -IC:\vvbuildq\paraview\src\paraview-build\VTK\Interaction\Style -IC:\vvbuildq\paraview\src\paraview\VTK\Interaction\Style -IC:\vvbuildq\paraview\src\paraview-build\VTK\Filters\Sources -IC:\vvbuildq\paraview\src\paraview\VTK\Filters\Sources -IC:\vvbuildq\paraview\src\paraview-build\VTK\Rendering\Core -IC:\vvbuildq\paraview\src\paraview\VTK\Rendering\Core -IC:\vvbuildq\paraview\src\paraview-build\VTK\Common\Color -IC:\vvbuildq\paraview\src\paraview\VTK\Common\Color -IC:\vvbuildq\paraview\src\paraview-build\VTK\Filters\Geometry -IC:\vvbuildq\paraview\src\paraview\VTK\Filters\Geometry -IC:\vvbuildq\paraview\src\paraview-build\VTK\Rendering\OpenGL -IC:\vvbuildq\paraview\src\paraview\VTK\Rendering\OpenGL -IC:\vvbuildq\paraview\src\paraview-build\VTK\Imaging\Hybrid -IC:\vvbuildq\paraview\src\paraview\VTK\Imaging\Hybrid -IC:\vvbuildq\paraview\src\paraview-build\VTK\IO\Image -IC:\vvbuildq\paraview\src\paraview\VTK\IO\Image -IC:\vvbuildq\paraview\src\paraview-build\VTK\Utilities\DICOMParser -IC:\vvbuildq\paraview\src\paraview\VTK\Ut [... cut ...]
c:\users\xrezv\downloads\veloview\velodynehdl\vtkDataPacket.h(95) : error C4430: missing type specifier - int assumed. Note: C++ does not support default-int

c:\users\xrezv\downloads\veloview\velodynehdl\vtkDataPacket.h(95) : error C4430: missing type specifier - int assumed. Note: C++ does not support default-int

c:\users\xrezv\downloads\veloview\velodynehdl\vtkDataPacket.h(101) : error C2146: syntax error : missing ';' before identifier 'blockIdentifier' 
...
```

it goes on for several more lines, I can upload whole log if it helps.
I will try to compile with VS2012 and see if that goes better.

Anarchon-github, 2017-12-25:
Thanks for the help with the Visual Studio, the problem was indeed in using earlier version.
When I used VS2012 x86 with same configuration as posted above, it gave me an error when linking PythonQT. I corrected it by using the superbuild version of Qt.

Here's the final configuration that I successfuly used for the build:

VS2012 x86 native tools command prompt,
cmake v.3.10 configured with cmake-gui with following options:
ninja generator (path to ninja exe in environment variables)
CMAKE_BUILD_TYPE = Release
CMAKE_INSTALL_PREFIX = c:\vvbuildq
ENABLE_qt = ON
ENABLE_veloview = ON
 
I also suggest updating the documentation to mention you need VS 11.0 instead of VS 9 as it is now.
Thanks once again!

closed 2017-12-25 by Anarchon-github