Kitware/VeloView #40: libprotobuf ERROR

fungb, 2018-02-10:
Ubuntu 16.04 LTS, running Veloview following a successful build gives the following error:

libprotobuf ERROR /veloview/veloview/build/paraview/src/paraview/ThirdParty/protobuf/vtkprotobuf/src/google/protobuf/descriptor_database.cc:57] File already exists in database: vtkPVMessage.proto
libprotobuf FATAL /veloview/veloview/build/paraview/src/paraview/ThirdParty/protobuf/vtkprotobuf/src/google/protobuf/descriptor.cc:860] CHECK failed: generated_database_->Add(encoded_file_descriptor, size): 
Aborted (core dumped)

bastienjacquet, 2018-02-15:
Hello,
This is unexpected, and does not happen on our 16.04.03 LTS machines.
Can you describe your setup, and do you have a system install of paraview or libprotobuf ?
Best,
Bastien Jacquet
VeloView lead developer

iravi89, 2019-08-29:
I got the same error, what steps are need to solve these errors