Kitware/VeloView #41: Building from source error

ssy248, 2018-02-10:
When I run : cmake -DENABLE_veloview:BOOL=ON ../SuperBuild

The following error shows up:

CMake Error at Projects/qt.cmake:17 (message):
  CMAKE_OSX_SYSROOT or CMAKE_OSX_ARCHITECTURES are not configured
Call Stack (most recent call first):
  CMakeLists.txt:123 (include)

-- Configuring incomplete, errors occurred!

bastienjacquet, 2018-02-15:
Hello,
I assume you are using a Mac and you want to compile your own Qt4.
Then you need to have Xcode installed, and point CMake to the Xcode sdk .
Running cmake-gui or ccmake would give you the ability to do this.
Elsewise, something along those lines should do it:
`cmake -DENABLE_veloview:BOOL=ON  -DCMAKE_OSX_ARCHITECTURES:STRING="x86_64" -DCMAKE_OSX_SYSROOT:STRING=/path/to/XcodeSDK/Developer/SDKs  ../SuperBuild
`
Best,
Bastien Jacquet
VeloView lead developer

ssy248, 2018-02-20:
Ok, I did that.

closed 2018-02-20 by ssy248