Kitware/VeloView #43: Veloview fails to launch after compiling with all enabled

rukie, 2018-02-18:
I get the following 
'error:
void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "" under id 22 
libprotobuf ERROR /home/rukie/build/paraview/src/paraview/ThirdParty/protobuf/vtkprotobuf/src/google/protobuf/descriptor_database.cc:57] File already exists in database: vtkPVMessage.proto
libprotobuf FATAL /home/rukie/build/paraview/src/paraview/ThirdParty/protobuf/vtkprotobuf/src/google/protobuf/descriptor.cc:860] CHECK failed: generated_database_->Add(encoded_file_descriptor, size): 
Aborted (core dumped)
'

Process
sudo apt-get install cmake-curses-gui build-essential libboost-all-dev libxt-dev libbz2-dev libqt4-dev qt4-default qt4-dev-tools zlib1g-dev software-properties-common
sudo apt-get update
sudo apt-get upgrade

cd
git clone git://public.kitware.com/VeloView.git
mkdir build
cd build
ccmake ~/VeloView/Superbuild
It starts with an empty cache, press c to configure
Enable all flags, boost, eigen, liblas, paraview, pcap, python, pythonqt, qt, veloview
press c

get the following
'error:
CMake Error at CMake/ParaViewModules.cmake:163 (set_property):
   set_property could not find CACHE variable ENABLE_png.  Perhaps it has not
   yet been created.
 Call Stack (most recent call first):
   CMakeLists.txt:133 (process_dependencies)

 CMake Error at CMake/ParaViewModules.cmake:188 (include):
   include could not find load file:

     png
 Call Stack (most recent call first):
   CMakeLists.txt:133 (process_dependencies)

 CMake Warning (dev) at /usr/share/cmake-3.5/Modules/ExternalProject.cmake:451 (message):
   value '' with no previous keyword in PVExternalProject_Add
 Call Stack (most recent call first):
   CMake/PVExternalProject.cmake:49 (_ep_parse_arguments)
   CMake/ParaViewModules.cmake:319 (PVExternalProject_Add)
   CMake/ParaViewModules.cmake:189 (add_external_project_internal)
   CMakeLists.txt:133 (process_dependencies)
 This warning is for project developers.  Use -Wno-dev to suppress it.

 CMake Error at /usr/share/cmake-3.5/Modules/ExternalProject.cmake:1915 (message):
   No download info given for 'png' and its source directory:

    /home/rukie/build/png/src/png

   is not an existing non-empty directory.  Please specify one of:

    * SOURCE_DIR with an existing non-empty directory
    * URL
    * GIT_REPOSITORY

Errors occurred during the last pass                  
'

press e to exit help
turn on Use_SYSTEM_PYTHON, 
no errors, press c,
turn on
USE_SYSTEM_BOOST, USE_SYSTEM_QT
press c
press g
export QT_SELECT=qt4
sudo make -j2

builds successfully
when trying to launch from install/bin/Veloview I get the error stated at the top

uname -r
4.10.0-28-generic
gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.6) 5.4.0 20160609
python --version
Python 2.7.12

This is pretty close to a fresh ubuntu install. I've added dependencies for ROS and libpcl-dev. 

I'd appreciate any help troubleshooting!

rukie, 2018-02-18:
libprotobuf-dev is already the newest version (2.6.1-1.3) is installed on the machine

and machine details:
H/W path         Device      Class          Description
=======================================================
                             system         HP EliteBook 8470p (C1U25AW#ABA)
/0                           bus            179B
/0/d                         memory         64KiB BIOS
/0/0                         processor      Intel(R) Core(TM) i5-3320M CPU @ 2.60GHz
/0/0/2                       memory         32KiB L1 cache
/0/0/3                       memory         256KiB L2 cache
/0/0/4                       memory         3MiB L3 cache
/0/1                         memory         32KiB L1 cache
/0/5                         memory         4GiB System Memory
/0/5/0                       memory         DIMM [empty]
/0/5/1                       memory         4GiB SODIMM DDR3 Synchronous 1600 MHz (0.6 ns)
/0/100                       bridge         3rd Gen Core processor DRAM Controller
/0/100/2                     display        3rd Gen Core processor Graphics Controller
/0/100/14                    bus            7 Series/C210 Series Chipset Family USB xHCI Host Controller
/0/100/14/0      usb3        bus            xHCI Host Controller
/0/100/14/1      usb4        bus            xHCI Host Controller
/0/100/16                    communication  7 Series/C210 Series Chipset Family MEI Controller #1
/0/100/16.3                  communication  7 Series/C210 Series Chipset Family KT Controller
/0/100/19        enp0s25     network        82579LM Gigabit Network Connection
/0/100/1a                    bus            7 Series/C210 Series Chipset Family USB Enhanced Host Controller #2
/0/100/1a/1      usb1        bus            EHCI Host Controller
/0/100/1a/1/1                bus            Integrated Rate Matching Hub
/0/100/1a/1/1/1              generic        VFS491
/0/100/1a/1/1/3              multimedia     HP HD Webcam [Fixed]
/0/100/1b                    multimedia     7 Series/C210 Series Chipset Family High Definition Audio Controller
/0/100/1c                    bridge         7 Series/C210 Series Chipset Family PCI Express Root Port 1
/0/100/1c.1                  bridge         7 Series/C210 Series Chipset Family PCI Express Root Port 2
/0/100/1c.2                  bridge         7 Series/C210 Series Chipset Family PCI Express Root Port 3
/0/100/1c.2/0                bus            IEEE 1394 Host Controller
/0/100/1c.2/0.1              generic        SD/MMC Host Controller
/0/100/1c.2/0.2              generic        Standard SD Host Controller
/0/100/1c.3                  bridge         7 Series/C210 Series Chipset Family PCI Express Root Port 4
/0/100/1c.3/0    wlo1        network        Centrino Advanced-N 6205 [Taylor Peak]
/0/100/1d                    bus            7 Series/C210 Series Chipset Family USB Enhanced Host Controller #1
/0/100/1d/1      usb2        bus            EHCI Host Controller
/0/100/1d/1/1                bus            Integrated Rate Matching Hub
/0/100/1d/1/1/6              generic        BCM20702A0
/0/100/1f                    bridge         QM77 Express Chipset LPC Controller
/0/100/1f.2                  storage        7 Series Chipset Family 6-port SATA Controller [AHCI mode]
/0/2             scsi0       storage        
/0/2/0.0.0       /dev/sda    disk           180GB INTEL SSDSC2BW18
/0/2/0.0.0/1     /dev/sda1   volume         487MiB Linux filesystem partition
/0/2/0.0.0/2     /dev/sda2   volume         167GiB Extended partition
/0/2/0.0.0/2/5   /dev/sda5   volume         167GiB Linux filesystem partition
/0/3             scsi1       storage        
/0/3/0.0.0       /dev/cdrom  disk           CDDVDW SN-208DB
/1                           power          CC06062

rukie, 2018-02-18:
I also have installed paraview and ROS (robot operating system)

rukie, 2018-02-19:
And while trying to install dependencies for the Point Cloud Library, its broken with the same error...

rukie, 2018-02-19:
Removing libprotobuf-dev and a system install of paraview seems to make it functional... 
If make -install will put Veloview into /usr/local/bin, why won't it put the compiled paraview there as well?

bastienjacquet, 2018-03-08:
Hello rukie,

I'm glad you solved the issue by yourself.
System-wide installs of libprotobuf does conflict with veloview and should be avoided for now.

vnn0804, 2019-02-17:
hey hi try 16.04 ubuntu. and build dir not required. follow the following steps

1. First, install the following dependencies using the apt-get commands
   git
   cmake-curses-gui
   build-essential
   libboost-all-dev
   libxt-dev
   libbz2-dev
   libqt4-dev
   qt4-default
   qt4-dev-tools
   zlib1g-dev
   libpcap-dev.
   export QT_SELECT=qt4
2. create project dir
   mkdir project
3. clone the veloview project
   git clone https://github.com/Kitware/VeloView.git
4. cmake build type following commands
   a) ccmake VeloView/Superbuild
   b) type c to configure
   c) page 1 for cmake
   CMAKE_BUILD_TYPE                *Release
   CMAKE_INSTALL_PREFIX            */home/icarus/project/
   ENABLE_boost                    *OFF
   ENABLE_eigen                    *OFF
   ENABLE_liblas                   *OFF
   ENABLE_paraview                 *OFF
   ENABLE_pcap                     *OFF
   ENABLE_python                   *OFF
   ENABLE_pythonqt                 *OFF
   ENABLE_qt                       *OFF
   ENABLE_veloview                 *ON
   qt_EXTRA_CONFIGURATION_OPTIONS  *

again type c (configure) and type e for exit

d) page 2 for cmake
USE_SYSTEM_boost *ON
USE_SYSTEM_pcap *ON
USE_SYSTEM_python *ON
USE_SYSTEM_qt *ON
CMAKE_BUILD_TYPE Release
CMAKE_INSTALL_PREFIX /home/icarus/project
ENABLE_veloview ON
qt_EXTRA_CONFIGURATION_OPTIONS

type c again type c and finally type g

1. run  cmake -DENABLE_veloview:BOOL=ON VeloView/Superbuild
2. run  make -j4 and make in two different terminal
3. finally open veloview application--  cd install/bin
   ./VeloView