Kitware/VeloView #44: GPRMC revision not yet supported

Chatmetaleux, 2018-02-21:
In some new GPS, it uses a new sentence for GPRMC trames :
$GPRMC,HHMMSS.SS,A,DDMM.MMM,N,DDDMM.MMM,W,Z.Z,Y.Y,DDMMYY,D.D,V,M,**NS***CC<CR><LF>
with 'Navigational status' field added. (info can be found [here](https://hemispheregnss.com/gnssreference/GPRMC.htm)) :

As result, Veloview can't parse this new information and so doesn't show the GPS position in 'Position' tab.
[image]

bastienjacquet, 2018-02-21:
Hello,

The handling of this new sentence could easily be added in vtkHDLPositionReader.cxx
In the spirit of open-source software, do you feel like providing a patch that would add this functionnality?
We would happily review and include your changes.

Best regards,
Bastien Jacquet
VeloView lead developer

Chatmetaleux, 2018-02-21:
Hello,

I would like to, but I need first to handle ROS concepts and make some
tests for our Velodyne...
See you soon !

Samuel.

Chatmetaleux, 2018-03-19:
I have troubles to add the new gprmc sentence...The new one's length is between 80 and 81 char instead of 73 char.
Do you have a nice way to detect the end position ? looking for 0x00 ?

This implies to modify vtkVelodyneHDLPositionReader::vtkInternal::ProcessHDLPacket() ligne 177.

Chatmetaleux, 2018-03-23:
Fix added into pull request #47...