Kitware/VeloView #51: How to use VeloView for SLAM

ratking82, 2018-05-11:
I was wondering how I can use VeloView for SLAM? I have a VLP16 Lidar and a XSens IMU. I did not find any instructions on how I can interface my IMU with VeloView. Also, I am curious how I can interface a separate GPS too. Thanks in advance!

etanx, 2019-04-19:
I was hoping they'd have this feature too! Frustrated crawling through codes all over GitHub trying to get SLAM (without GPS and IMU) to work on .pcap files. But recently in Feb I heard news that Veloview 4.0 will have in-built SLAM so hopefully that happens soon....!

PierreGuilbertF, 2019-04-19:
Hello,

To run the SLAM algorithm you need to:

- Enable the Pipleline Browser view: View > Pipeline Browser
- Enable the Properties View view: View > Properties

Once a .pcap is opened you will be able to see its instantiation within the pipeline browser view. Select the "Calibration" source and then press "ctrl + space" and type SLAM. You should be able to select the filter and instantiate it (see screenshot).

[image]

There is two filters (using the same SLAM algorithm), an online one that shows the SLAM being computed. Using this mode, you can stop the slam earlier if there is a drift or issues. An offline one where you just select the frame interval you want to process.

Once you selected offline / online slam a window will pop to select the "Point Cloud" source and the "Calibration" source:

[image]

Finally, you will be able to see the properties of the slam algorithm:
[image]

The default settled parameters are a good compromise to make the SLAM working in outdoor urban area, indoor scene and poor geometric scene (forest recorded from UAV, glades, career, ...).

Choosing the online SLAM improve the understanding of what is being computed. You can display the maps computed so far, the trajectory and the position of the last frame:

[image]

Once the you're done processing your frames, you can export the results choosing the data you want to export and using "ctrl+s"

Regards,
Pierre Guilbert

etanx, 2019-04-20:
But if building (on Windows 10) with cmake [following these instructions](https://github.com/Kitware/VeloView/blob/master/Documentation/VeloView_Developer_Guide.md#superbuild-overview), where/how would one specify the compile options?

PierreGuilbertF, 2019-04-23:
By adding the ENABLE_pcl and ENABLE_ceres:

`cmake <work-directory>\Veloview-source\Superbuild -GNinja -DCMAKE_BUILD_TYPE=Release -DUSE_SYSTEM_qt5=True -DQt5_DIR="C:/Qt/Qt5.10.0/5.10.0/msvc2015_64/lib/cmake/Qt5" -DENABLE_pcl=True -DENABLE_ceres=True`

This command will configure the superbuild so that pcl and ceres sources will be downloaded and compiled if required. Once pcl and ceres are ready to be used, you will have to open the VeloView project cmake and enable pcl and ceres.

This is because

- Enabling pcl and ceres in the superbuild project will command the project to download and compiled the dependencies

- Enabling pcl and ceres in the VeloView project will add the dependencies to the project and unlock the features that require pcl and ceres

etanx, 2019-04-25:
That command was very helpful! I now see pcl and ceres files in the build folder, and am able to locate the Pipeline Browser. However, I did not find SLAM filters while searching, and I'm guessing this is because I have not enable pcl and ceres in the Veloview cmake. I poked around some cmake files in `C:\VeloView-build\common-superbuild\veloview\build` and searched up help on cmake but couldn't figure out. 

Where can I find the Veloview project cmake and how should it be changed? Thanks for taking time to provide guidance!

PierreGuilbertF, 2019-04-26:
I suggest that you open a VS2015 cmd terminal in folder "${PATH_TO_BUILD}\common-superbuild\veloview\build" and open cmake-gui using " cmake-gui . " . Calling cmake-gui with option "." will open the VeloView cmake project.
You should then be able to see the options ENABLE_pcl and ENABLE_ceres. Once you have enabled them, you just need to recompile VeloView and install it. To do that, close cmake-gui, and use the same VS2015 cmd prompt at the same location and type "ninja install"

Tell me if you have any problem doing this

Regards,
Pierre Guilbert

etanx, 2019-04-26:
Now SLAm features are properly enabled! When trying to run SLAM on a .pcap, going to `Pipeline browser > Calibration > SLAM online > Point Cloud input port > Frame (Input Dialog) > 'OK'`, an error shows up:
```
ERROR: In C:\VeloView-build\common-superbuild\paraview\src\VTK\Common\ExecutionModel\vtkDemandDrivenPipeline.cxx, line 710
vtkPVCompositeDataPipeline (000001E0E803BDD0): Input port 1 of algorithm vtkSlam(000001E0E1F538B0) has 0 connections but is not optional.
```
Properties of SLAM algorithm can be found by typing in the search bar. But if I hit 'Apply' in the properties bar, another section of error appears in addition to the codes above"
```
RROR: In C:\VeloView-build\common-superbuild\paraview\src\VTK\Common\ExecutionModel\vtkExecutive.cxx, line 784
vtkCompositeDataPipeline (000002430A21CA60): Algorithm vtkPVGeometryFilter(000002430A0E09E0) returned failure for request: vtkInformation (000002430A2916A0)
  Debug: Off
  Modified Time: 13621708
  Reference Count: 1
  Registered Events: (none)
  Request: REQUEST_DATA_OBJECT
  ALGORITHM_AFTER_FORWARD: 1
  FORWARD_DIRECTION: 0
  FROM_OUTPUT_PORT: 0
```
[image]

PierreGuilbertF, 2019-04-26:
We are almost there! :)
The graphical interface is a bit tricky here.

The error you got is telling you that the algorithm requires two inputs (one is the pointcloud and the second one is the calibration table) and that the input 1 (the calibration table) is empty.

You have to do:
Pipeline browser > Calibration > SLAM online

Once the dialog popped up:
Point Cloud input port > Frame (Input Dialog)
Calibration > Calibration > ok

And then it should works.

Let me know if it works

Regards,
Pierre Guilbert

PierreGuilbertF, 2019-04-26:
One last thing, by default the speed is settled to x1 meaning that the software will drop some frames if the algorithm does not run fast enought. Of course, frame dropping will affect the SLAM and reduce its accuracy / robustness.
To avoid that you can select "all frames" option so that the SLAM algorithm will processes all frames without dropping.

This remark only concerns the online SLAM.
The SLAM will be updated soon with improvements on the speed (~60% faster) and the accuracy

etanx, 2019-04-28:
It finally works! Tried online and offlien SLAM with with data from VLP-32c. Even tried to live stream sensor input _and_ SLAM at the same time - software struggled a bit (sometimes crashes) but it was able to do that.

I have last three questions:

1. To select "all frames" does that refer to the trailing frames?
2. What's the best way to export the completed point coud ? I tried _Advaced > File > Save Data_ and chose the .ply format to load in Paraview but nothing was shown (didn't save correctly?)
3. I noticed the point clouds were white and did not display the coloured intensity values. Is there a feature to enable the color intensity in final point cloud?

PierreGuilbertF, 2019-04-29:
Hi,

> To select "all frames" does that refer to the trailing frames?

No, selecting the "all frames" option prevents the software from dropping frame to achieve real-time display. Let's say you are processing a VLP-16 at 10Hz. If it takes 500ms to estimate the motion of the sensor within a frame, then, if you are in x1 (realtime), at the end of the motion estimation the software will drop 4 frames and process the 5th frame. Selecitng "All frames" will make sure that the SLAM will process all the frame and be more accurate.

> What's the best way to export the completed point coud ? I tried Advaced > File > Save Data and chose the .ply format to load in Paraview but nothing was shown (didn't save correctly?)

Actually, what is being displayed are the "planars" and "edges" keypoints that have been used to estimate the motion of the sensor. For now, you can export them by selecting the pointcloud in the pipeline browser ans then "ctrl + s". You can also export the estimated poses trajectory (with the .poses file format associated). To export the all frames processed, you need to:
- Instanciate a "Transforms Applier" that takes two inputs, the pointcloud generator (data) and the trajectory (the one estimated by the slam for instance).
- "ctrl + s" export as .las

> I noticed the point clouds were white and did not display the coloured intensity values. Is there a feature to enable the color intensity in final point cloud?

The "edge" and "planar" map displayed don't have any information associated to the points. However, you can use the trailing frames memchanism + transform applier using the estimated poses trajectory to have a dense registered point cloud. Doing this, you will have all usual information per points (intensity, time, ...)

Regards,
Pierre Guilbert
-

Matioupi, 2019-04-30:
Hello, I'm strictly following the instructions given here, but I cannot get SLAM module accessible.

my CMakeCache.txt has this :

`//ON
ENABLE_Ceres:BOOL=OFF

//ON
ENABLE_PCL:BOOL=OFF

//No help, variable specified on the command line.
ENABLE_ceres:BOOL=ON

//No help, variable specified on the command line.
ENABLE_pcl:BOOL=ON`

Could it be some case sensitivity issue on the compile switch given though example command line : https://github.com/Kitware/VeloView/issues/51#issuecomment-485677263

Regards

etanx, 2019-05-01:
Strange, my CMakeCache.txt only has the first two. Did you get to use cmake-gui to enable pcl and ceres (https://github.com/Kitware/VeloView/issues/51#issuecomment-487028482) after the first installation as well?

etanx, 2019-05-01:
> Instanciate a "Transforms Applier" that takes two inputs, the pointcloud generator (data) and the trajectory (the one estimated by the slam for instance).

@PierreGuilbertF I tried selecting trailing frames as the pointcloud input, since I could not select 'SlamOffline1' which I guess is the whole pointcloud. Selected Trajectory for the trajectory input. But the error (below) seems to suggest that these are not the right inputs for the Transforms Applier. 

[image]

> "ctrl + s" export as .las

Somehow the options do not show a .las format, and forcing a las format does not work. Any suggestions?

[image]

```
ERROR: In C:\VeloView-source\VelodyneHDL\IO\vtkTemporalTransformsWriter.cxx, line 43
vtkTemporalTransformsWriter (00000246F69ABE70): Input is not a vtkTemporalTransforms

ERROR: In C:\VeloView-build\common-superbuild\paraview\src\VTK\Common\ExecutionModel\vtkExecutive.cxx, line 784
vtkPVCompositeDataPipeline (00000246984E29E0): Algorithm vtkTemporalTransformsWriter(00000246F69ABE70) returned failure for request: vtkInformation (00000246F8FBC850)
  Debug: Off
  Modified Time: 558525993
  Reference Count: 1
  Registered Events: (none)
  Request: REQUEST_DATA
  FROM_OUTPUT_PORT: -1
  ALGORITHM_AFTER_FORWARD: 1
  FORWARD_DIRECTION: 0
```

PierreGuilbertF, 2019-05-02:
@Matioupi Can you check that the superbuild has downloaded and compiles ceres and pcl ? To do that, you can go in the build folder > common-superbuild > pcl / ceres. If the projects have been downloaded and compiled it means that the problem comes from the VeloView CMake project (and not the superbuild one).

@etanx I noticed that you are using the SLAM on indoor data. While the default parameters have been settled to provide a SLAM working in most situations, you can improve the results by reducing the voxel grid leaf size of the maps (both edges and planar). For example, using 0.15 (15cm) for the edges map and 0.25 (25 cm) for the planar should improve a lot the results. Moreover, you can reduce the "Min Dist To Sensor" parameter from 3 meters to 1 meter.

To use the trailing frame + transform applier you should:

- Select the trailing frame and set the desired number of trailing frame (it will display the frames [|current_frame - Ntrailingframe, current_frame|]). Click on apply. You should now see all the frames aggregated in a non-sense way (all points being displayed using their coordinates of the reference frame attached to the sensor at the time of acquisition).

- Instantiate a "Transform Applier" with point cloud <-> the output of the Extract Surface and trajectory <-> the one estimated by the SLAM. 

From what I can see, the Transforms Applier filter is complaining about the "Trajectory" input. Have you match the Transforms Applier "Trajectory" input with the SLAM "Trajectory" output ? If yes, it may be a bug with the SlamOffline (we usually use the online one).

Note 1: once the SLAM has been ran, you can export the Trajectory as .poses to avoid running the SLAM again (selecting the Trajectory and ctrl + s). Then, to load the trajectory you can drag and drop the .poses file.

Note 2: Be sure that all the "filters" (elements of the pipeline browser) have been applied. To check that, you can select the filter and check if the "Apply" button is enabled or disabled. It can be required when loading a filter. For example, if you load a .poses file, you must click on apply to activate it.

Regards,
Pierre Guilbert

Matioupi, 2019-05-02:
Yes, ceres and pcl are downloaded and build.
At the step where I enable ceres and pcl from cmake-gui . after enabling the 2 tickboxes, am I supposed to press the "configure" button before quitting the GUI and running ninja install again ? (that's what I did)

etanx, 2019-05-06:
@Matioupi after enabling the 2 tickboxes I pressed 'Generate' to create new files (if I remember correctly), then ran `ninja install` to install those new files.

Matioupi, 2019-05-08:
Solved... after pressing ctrl+ space, I was expecting to see the slam options in the box "out of the box"....
After typing SLAM (and not clicked "SLAM") it seems all good !
Regards and thanks for the help !

hackora, 2019-05-09:
Hello,
I'm currently trying to use SLAM and it seems that everytime I run it I get this red message which I dont know if it's an error, it also shows up with all the frames that are being processed.
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[image]

Is it impacting the output in anyway? If so how do I fix this?
Thanks

PierreGuilbertF, 2019-05-09:
Hello Hackora,

This message is an artifact of next features to come within the SLAM algorithm. It doesn't impact the result of the SLAM and there is nothing to fix for now.

It is due to an attempt of building a kdtree upon the "blob keypoints" environment map that is not implemented yet.

Regards,
Pierre Guilbert

hackora, 2019-05-09:
@PierreGuilbertF, Great! thank you for the quick response ^^

cwg18, 2019-05-19:
@PierreGuilbertF , is there a guide for doing this on Ubuntu 16.04? The moment I enable pcl I am unable to compile Veloview.

Matioupi, 2019-05-21:
When exporting the data, I end up with on .ply / .csv per frame, is there a way to export the result in a unique file ? Is there a way to export to .las/.laz directly from veloview ?

cwg18, 2019-05-24:
Hi @PierreGuilbertF,

Sorry about the previous message. I managed to compile on Ubuntu 16.04. I did it by using ccmake and forcing it to not use system versions of most things. I managed to compile it with the ceres and pcl options enabled. Once the program has finished compiling, I loaded a pcap file. When I selected calibration and pressed ctrl+enter, I cannot find slam in the list. Is there something I am missing?

PierreGuilbertF, 2019-05-24:
@cwg18 Are you sure you have enabled Ceres and PCL both in the superbuild cmake AND in the VeloView cmake project ?

- Enabling those dependencies on the superbuild will configure the superbuild to download and build the libs. You can check that everything went right by looking at `build > install > lib` and checking if the libraries have been compiled

- Enabling the dependencies in the VeloView project will configure VeloView to build the algorithms depending on those libs (SLAM, Camera - Lidar calibration, ...)

@Matioupi 
Yes, you can directly export the data in a .las / .vtp file from VeloView. To do that, you must have:

- A pointcloud source

- A position / orientation over time source (the result of the SLAM for instance)

You must instantiate a `Temporal Transform Applier` (see screenshot below). To export as a .las file, you can press "ctrl + s" and select export as .las file.

[image]

Once the Temporal Transform has been applied you will notice that the pointcloud is now expressed in a fixed reference frame (the one attached to the sensor at the beginning of the SLAM processing). Thus, you can use a trailing frame to aggregate the frames:

[image]
 And finally, it is possible to press "ctrl + s" and choose .vtp export format

Regards,
Pierre Guilbert

cwg18, 2019-05-24:
Hi @PierreGuilbertF,

I am not sure what to look for but if you look at the screenshots, in the make process pcl was compiled with no issues and I see some libpcl files in the build > install > lib folder. What should I see if ceres is compiled correctly?

[image]
[image]
[image]

PierreGuilbertF, 2019-05-24:
Hi @cwg18 

Perfect regarding PCL. Ceres is a header only library, to check if the installation went fine you should check if the folder `build > install > include > ceres` is populated.

To check if you have enabled Ceres and PCL in the VeloView project you should open a terminal in the folder `build > common-superbuild > veloview > build`  and enter the command line `cmake-gui .` The GUI will show you if the libraries are enabled:

[image]

Regards,
Pierre Guilbert

cwg18, 2019-05-24:
Hi @PierreGuilbertF,

I checked the ceres folder and it is populated.

In the cmake-gui, the ceres and pcl is enabled. Whats next?
[image]

Sorry about this. I've been banging walls for a few weeks now so getting a little frustrated.

PierreGuilbertF, 2019-05-24:
@cwg18 

According to your screenshot you have opened the superbuild cmake project (check the source code path). You need to open the VeloView project cmake project.

To do that, open a terminal in `build > common-superbuild > veloview > build` and enter the command line `cmake-gui .`  <--- whatch out the point `.` required in the command line. It should open the VeloView cmake project, you can check that by looking the source code path

Regards,
Pierre Guilbert

Matioupi, 2019-05-24:
@cwg18 :

after hitting ctrl+space to display the dialog box,you need to enter the first letters of word "SLAM" from the keyboard. I was missing that step the first times. Maybe it'sthe same for you.
It's the same to access the ICP properties or min distance, you (at least I) have to type the ICP / dist word on the propertie filter input box.

cwg18, 2019-05-27:
@PierreGuilbertF Sorry for the delay. Had something to do for the weekend. I have followed your instructions. In this one, enable pcl and ceres are disabled. What do I do next?

@Matioupi if you look at the screenshot above, I have typed at least sla. It doesn't appear.

arier777rtx, 2019-06-18:
> That command was very helpful! I now see pcl and ceres files in the build folder, and am able to locate the Pipeline Browser. However, I did not find SLAM filters while searching, and I'm guessing [...]

I have problem during build install see error below:

Submodule path 'ThirdParty/IceT/vtkicet': checked out '0d08b037f3fa8b6e1a82ed2beb84e024c06f895f'
Submodule path 'ThirdParty/QtTesting/vtkqttesting': checked out 'c44b32fdea827be737e8c2f5608ffbc2e3bd08b2'
Submodule path 'ThirdParty/cinema/vtkcinemapython': checked out '77c759a81b7f2536cacc9c1a0875eb2c1f5f22bd'
Submodule path 'ThirdParty/protobuf/vtkprotobuf': checked out '3637bc9efc2c29c54ee907e4341e39127599c4ec'
Submodule path 'Utilities/VisItBridge': checked out '649a217583d31703eff3d2f82bbd195756c8a551'
Submodule path 'VTK': checked out 'd507749b06954dca021769de9f5008f7228a9dcf'
ninja: build stopped: subcommand failed.

C:\VeloView-build>

kindly need your suggest & advices thank you

mckenziebl, 2019-06-23:
Hello!
I'm working on VeloView with SLAMing properties on Ubuntu 16.04. 
I have gotten to the point where ceres and pcl are made and enabled on the cmake-gui . However, when I open the application I can not find the pipeline browser option in the Views folder tab. 
I do not have any errors from the system, but I am not sure if I am compiling VeloView correctly with the changes as I am brand new to Linux systems and commands. 

Any help is greatly appreciated! Thank you

cwg18, 2019-06-24:
@mckenziebl I think you ran into a similar problem like what I had.

Trychecking the cmake for veloview itself. You can refer to https://github.com/Kitware/VeloView/issues/51#issuecomment-495572925

PierreGuilbertF, 2019-06-24:
@arier777rtx 
Since you're using ninja and according to the path example you provided I am assuming you are trying to build VeloView on Windows.
I tried this morning and I was not able to reproduce your error. Please, make sure that:

- The submodules have been initialized (using git bash on the VeloView sources folder type: `git submodule update --init --recursive Superbuild`)

- You configuring the ninja project typing, from the VS2015 cross tools x84 x64 prompt, the command line: `cmake <work-directory>\Veloview-source\Superbuild -GNinja -DCMAKE_BUILD_TYPE=Release -DUSE_SYSTEM_qt5=True -DQt5_DIR="C:/Qt/Qt5.10.0/5.10.0/msvc2015_64/lib/cmake/Qt5" -DENABLE_pcl=True -DENABLE_ceres=True`

@mckenziebl
First of all, if you are building VeloView on windows and running the program using intel HD integrated GPU you might have a graphical bug where a white band is hiding the VeloView menu-bar (see below).

[image]

If you have such a bug, I recommend to launch the program using an other GPU. To enable the pipeline browser, you should first enable the advanced feature in the Help menu (see below)

[image]

Finally, once the advanced feature is enabled, you can open the pipeline browser in the Views menu (see below)

[image]

Regards,
Pierre Guilbert

mckenziebl, 2019-06-25:
@PierreGuilbertF 
Thank you so much! I now have Pipeline enable and I am able to SLAM.

PierreGuilbertF, 2019-06-25:
@mckenziebl 
You're welcome!
Depending on the kind of data you have (indoor, outdoor, urban, nature, ...) the parameters of the SLAM might need to be tuned. Let me know if you need some help on this part.
The default parameters have been chosen to work well with urban outdoor scene with low-speed vehicle and 600 RPM sensor.
To be able to change the SLAM parameters, you need to enable the "Properties" view and click on advanced-feature (gear icon)

Regards,
Pierre Guilbert

mckenziebl, 2019-06-26:
@PierreGuilbertF 
I've been playing with the parameters a bit already. We're working outside carrying the puck on foot at an archeological dig. We're working in areas of 10x10 meters. Right now our RPM is set at 300, but we will probably be lowering that. A lot of our work seems to just need to be cut away and we're having difficulties getting the frames to align when we rotate slowly. Any advice would be great!

RBarron01, 2019-06-28:
Hi, I also am having trouble saving as an .las file. I have applied the temporal transform applier to the .poses file I saved of the trajectory but don't see the option to save as .las. Any pointers would be great!
[image]

Laith2019, 2019-07-03:
@PierreGuilbertF
I have installed veloview software and found it does not have the following lists (Sources, Filters, Views, Advance) as you showed. So could you please advise me how to get these lists and how to get SLAM in veloview. I am very new in this field?
I have attached to you screen shoot of the installed veloview
[image]

Thank you in advance

etanx, 2019-07-04:
@Laith2019 did you build Veloview from source? Try out the instructions in this Veloview SLAM Guide below, or read this thread's history from [this comment](https://github.com/Kitware/VeloView/issues/51#issuecomment-485677263) onwards:

[](https://github.com/etanx/VeloView/blob/master/Documentation/How-To-SLAM-With-Veloview.md)

Laith2019, 2019-07-04:
Thanks @etanx 
I have this error when doing step 6, I could not fix it. could you please help me?

[image]

Laith2019, 2019-07-05:
Can anyone help me please, I am stuck in this problem, and tried many ways but nothing fixed the problem.
I followed the steps in https://github.com/Kitware/VeloView/blob/master/Documentation/VeloView_Developer_Guide.md#configure-build
Until the step 6, I got the following error saying that
Prebuilt Python binaries for Windows 32 bit are not provided.
Call Stack (most recent call first):
common-superbuild/CMakeLists.txt:130 (include)
Configuring incomplete, errors occurred!

PierreGuilbertF, 2019-07-05:
@etanx 
Thanks a lot for your help and for this SLAM Guide, that helps a lot!
https://github.com/etanx/VeloView/blob/master/Documentation/How-To-SLAM-With-Veloview.md

@mckenziebl 
We have worked a lot with outdoor archeological environment geometry. While the parameters usually need to be adapted, we usually managed to make it work. Unfortunately, I can't provide you advises without having a look at the data. If your data are not sensitives you can send me a sample via WeTransfer and I will see what I can do.

@Laith2019 
You will need to completely clean your build directory (I suggest you to remove and recreate it). From the screenshot you sent it seems that you did not use the "VS2015 x86 **x64 Cross Tools** Command Prompt". I know it is tricky, but it only works with this specific command prompt. Then, you can redo what you did and it should work.

Let me know :)

Regards,
Pierre Guilbert

Laith2019, 2019-07-05:
Thanks @PierreGuilbertF 
But how get this command prompt "VS2015 x86 x64 Cross Tools Command Prompt"
I installed the vs2015 express and there was no command prompt,

PierreGuilbertF, 2019-07-05:
@Laith2019 
You can use the windows app research tool and type "vs2015" and windows should be able to propose you the different command prompts. However, I noticed that some visual studio 2015 installation failed to reference the prompts.

In this case, you can go directly on the folder of the application:
`C:\ProgramData\Microsoft\Windows\Start Menu\Programs\Visual Studio 2015\Visual Studio Tools\Windows Desktop Command Prompts`

Laith2019, 2019-07-05:
Thanks @PierreGuilbertF 

Now I found it and used it, but I got the following error, any clues on how to fix it?
Sorry for taking your time

[image]

arier777rtx, 2019-07-11:
> @arier777rtx
> Since you're using ninja and according to the path example you provided I am assuming you are trying to build VeloView on Windows.
> I tried this morning and I was not able to reproduce [...]

Hi Pierre,

Im get same error, kindly need your advices. thanks

[image]

Laith2019, 2019-07-11:
Can anyone help me please,
I am struggling  with this issue:
https://github.com/Kitware/VeloView/issues/51#issuecomment-508699700

Kind regards,

arier777rtx, 2019-07-13:
@PierreGuilbertF 
Hi Pierre,

I'm still get error, kindly need your advices see below :

[ 59%] Completed 'boost'
[ 59%] Built target boost
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
@ubuntu:~/veloview/veloview-build$

minifacture, 2019-08-06:
@PierreGuilbertF

Hello Pierre,

Thank you very much for your tutorial and answers for SLAM for Veloview. I have successfully built, installed and ran SLAM with data collected with a backpack mounted VLP-16. Now I am playing with the parameters and trying to understand them which I have a little idea of what each parameter does. 

I am new to SLAM and especially without IMU or GPS so lack of reference state makes me curious about things like below:

1. Maybe it is collected with a backpack the transform applier view always shows something like below, not like pretty and aligned view like yours. Is it normal? 
[image]

2. When I export the pointcloud by selecting Blob+Edge+Planar+Transform applier with .vtk (no .las in the drop down menu no matter how I select those). The exported pointcloud (.vtk) is really small (7k for the example image below) and the points look sparse. I imagined the points would be much denser and larger in file size.
[image]

3. I didn't have IMU but I had GPS locked when I recorded pcap in question above. I tried to export the position as .pose but when I tried to import GPS position, and error message said it is not a correct .pose file. Is there a way to align it with GPS location?

Thank you much for your time!
H

zzq1016, 2019-09-10:
Hello,  @PierreGuilbertF , 
could you give a guide、tutor or interface about object recognition?
 Thanks for any help!

tehkillerbee, 2019-10-27:
Hello @PierreGuilbertF , 

I have managed to compile VeloView with SLAM support and so far, everything appears to be working. When using the Temporal Transform Applier with the data Frames + SLAM trajectory, I get an output that appears as expected. But as soon as I enable Trailing Frames to generate the point cloud for .las export, VeloView crashes as soon as I click "Apply". No error message or anything. I dont know whats going wrong. Can you describe this step again, please?

Another question. How can I import IMU/GNSS data from an external file into VeloView? I would like to use these values to georeference my LiDAR point cloud.

Thank you!

nizovskiyka, 2020-01-15:
Hello,
I have managed to do everything described above with VeloView, but I can not make it use GPS data from the receiver for georeferencing. The manual says that I need to setup the gps receiver position and orientation with respect to lidar sensor, but this gives no result. Am I missing anything?
Thanks in advance!

EnzoMeertens, 2020-02-06:
> Hello @PierreGuilbertF ,
> 
> I have managed to compile VeloView with SLAM support (on Windows) and so far, everything appears to be working when I am processing a dataset acquired with a VLP-16. When using the Temporal Transform Applier with the data Frames + SLAM trajectory, I get an output that appears as expected. But when I enable Trailing Frames to generate the point cloud for .las export, VeloView crashes as soon as I click "Apply". No error message or anything. I don't know what's going wrong. Can you explain this step in more detail?
> 
> Another question. How can I import IMU/GNSS data from an external file into VeloView? I would like to use these values to georeference my LiDAR point cloud.
> 
> Thank you!

@PierreGuilbertF I seem to have the same issue. As soon as I apply the trailing frame to the Temporal Transform Applier, no matter what number of trailing frames I specify, VeloView crashes.

Is there perhaps another way to export as .LAS file? Or is there a way to prevent this crash?

kilocharlievictor, 2020-07-10:
Pierre,
I have read through most of the inputs on this feed. I am working with VLP16 and have been able to build the Veloview on x86 , with Ubuntu1804. (I did not enable PCL or Ceres).
1)Is the SLAM Support through PCL and Ceres also proven and enabled on Linux. Most of the discussions i have seen are on Windows with MSVC?
2)Is the Veloview build supported on NVIDIA GPU(aarch64 architecture), so that it is possible to run it on Nano/TX2 etc NVIDIA boards? Is there any cross-compile toolchain supporting this?
Please advise.
Thank you for the time and support.
Regards.

RBarron01, 2020-07-10:
@kilocharlievictor I have used Ubuntu 18.04 with this system and did finally get the issues sorted out. 1) You do need to enable PCL and CERES on a Linux box — and it is supported and I can confirm it works. 2) I also have a Nvidia GPU so that should be fine.
I hope that helps but let me know if you’re still having trouble!

kilocharlievictor, 2020-07-10:
RBarron01: 
1)Does that mean Nvidia GPU on the laptop is ESSENTIAL or is it OPTIONAL? If i do not have a GPU on my x86 Linux host , will this standalone simulator still work?
2)Is this simulator targetted for only amd64 _x86 or can it run on ARMv8/aarch64 (i.e. Nvidia Nano/Jetson Dev boards)?
Please advise.

kilocharlievictor, 2020-07-11:
@PierreGuilbertF ,
I have installed the Veloview from source and build it with PCL and  CERES enabled. 

The build goes through. I am on x86 amd64 host, with Linux 1804 and with NO Nvidia GPU on my laptop(latitude E6530). After enabling the adv features and Pipeline browser, when i choose calibration source, I dont see any SLAM options as mentioned in your snapshots in previous post. Attached is what I see in the list of filters for calibration source when I search for SLAM in that list.(No online or offline slam)
[image]

[image]

Can you please look into this and point potential reasons based on my configuration here.
Please advise.
Regards.
Kunal Vora

RBarron01, 2020-07-13:
@kilocharlievictor great question and honestly a little outside my area but I’ll take a stab at it. I know you don’t have to have a nvidia gpu. As for the arch type, since you’ve successfully built it, I think it should be fine (I think you’d get a make error if the arch type was incompatible). To be honest, I haven’t worked with Nvidia Nano or Jetson Dev boards. I think the errors you’re getting are software errors not hardware. 
We had some similar issues getting slam to show up. Did you enable it in the cmake GUI and during the build? Here’re the steps we followed for getting slam to show up:
[image]
[image]
[image]

Just want to make sure we’re covering our bases! (Also sorry about the slow response time)

[10 more replies]