Kitware/VeloView #52: VeloView SuperBuild fails

Chris-Burnette, 2018-05-29:
Trying to build VeloView following the Developer Guide's SuperBuild but am getting stuck when it builds ParaView. It reports that vtkOpenGLBufferObject,h isn't available. Looks like maybe the right include folder isn't being used for PointCloudPlugin. I've tried updating that but then am getting another error with vtkRenderingOpenGL2Module.h not being found (included in vtkOpenGLBufferObject,h). In looking at the repo, I don't see that file anywhere. I'm using VS 2012 (as per the build instructions). 

VeloView\build-x86\paraview\src\paraview\Plugins\PointCloudRepresentation\vtkPointCloudMapper.cxx(27): fatal error C1083: Cannot open include file: 'vtkOpenGLBufferObject.h': No such file or directory [C:\development\projects\VeloView\build-x86\paraview\src\paraview-build\Plugins\PointCloudRepresentation\PointCloudPlugin.vcxproj]

buaafish, 2018-06-07:
I failed too.

9>D:\code\VeloView\Build\paraview\src\paraview\Plugins\PointCloudRepresentation\vtkPointCloudMapper.cxx(27): fatal error C1083: Cannot open include file:“vtkOpenGLBufferObject.h”: No such file or directory [D:\code\VeloView\Build\paraview\src\paraview-build\Plugins\PointCloudRepresentation\PointCloudPlugin.vcxproj]

jesselmk, 2018-06-07:
Same problem.

fatal error C1083: Cannot open include file:“vtkOpenGLBufferObject.h”: No such file or directory [D:\code\VeloView\Build\paraview\src\paraview-build\Plugins\PointCloudRepresentation\PointCloudPlugin.vcxproj]

bastienjacquet, 2018-06-08:
As a result of improving rendering speed, our Paraview fork is currently broken if using the OpenGL rendering backend. 

This should be resolved soon, but meanwhile you have two options to make it compile:
- select OpenGL2 rendering backend by adding     `-DVTK_RENDERING_BACKEND:STRING=OpenGL2` to [Superbuild/Projects/paraview.cmake](https://github.com/Kitware/VeloView/blob/cbdbe68cede38caf5d5dbf2d14dd9519a51a6024/Superbuild/Projects/paraview.cmake#L50)
- ignore our rendering speed improvement in Paraview fork  by setting `Paraview GIT_TAG` in [Superbuild/versions.cmake L42](https://github.com/Kitware/VeloView/blob/cbdbe68cede38caf5d5dbf2d14dd9519a51a6024/Superbuild/versions.cmake#L42) to `GIT_TAG 92985c920abef7f6014b8035a82fa27981ea3366`

arier777rtx, 2019-06-18:
> As a result of improving rendering speed, our Paraview fork is currently broken if using the OpenGL rendering backend.
>
> This should be resolved soon, but meanwhile you have two options to make it [...]

I have also issue build veloview, capture below :

Submodule path 'ThirdParty/IceT/vtkicet': checked out '0d08b037f3fa8b6e1a82ed2beb84e024c06f895f'
Submodule path 'ThirdParty/QtTesting/vtkqttesting': checked out 'c44b32fdea827be737e8c2f5608ffbc2e3bd08b2'
Submodule path 'ThirdParty/cinema/vtkcinemapython': checked out '77c759a81b7f2536cacc9c1a0875eb2c1f5f22bd'
Submodule path 'ThirdParty/protobuf/vtkprotobuf': checked out '3637bc9efc2c29c54ee907e4341e39127599c4ec'
Submodule path 'Utilities/VisItBridge': checked out '649a217583d31703eff3d2f82bbd195756c8a551'
Submodule path 'VTK': checked out 'd507749b06954dca021769de9f5008f7228a9dcf'
ninja: build stopped: subcommand failed.

C:\VeloView-build>

kindly advices and suggestion, thank you