Kitware/VeloView #55: Fail of paraview compilation

sergiocasaspastor, 2018-07-05:
Hi! I'm not able to install VeloView in linux due to an error in the build of paraview. I show you my error given in terminal. I follow all the instructions given in the manual for installing in linux. I can't find which is the error. 

[ 75%] Performing build step for 'paraview'
Scanning dependencies of target ParaViewData
Scanning dependencies of target VTKData
Scanning dependencies of target Pygments
[  0%] Built target VTKData
Scanning dependencies of target vtksys
Scanning dependencies of target vtkWrappingTools
[  0%] Copying files for Python package 'pygments'
[  0%] Built target ParaViewData
[  0%] Building C object VTK/Wrapping/Tools/CMakeFiles/vtkWrappingTools.dir/vtkParse.tab.c.o
Scanning dependencies of target vtkalglib
[  0%] Building C object VTK/Utilities/KWSys/vtksys/CMakeFiles/vtksys.dir/ProcessUNIX.c.o
Scanning dependencies of target vtkDICOMParser
[  0%] Building CXX object VTK/ThirdParty/alglib/CMakeFiles/vtkalglib.dir/ap.cpp.o
[  0%] Compiling Python package 'pygments'
[  0%] Building CXX object VTK/Utilities/DICOMParser/CMakeFiles/vtkDICOMParser.dir/DICOMFile.cxx.o
Listing '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments'...
Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/__init__.py'...
***   File "/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/__init__.py", line 46
    except TypeError, err:
                    ^
SyntaxError: invalid syntax

Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/cmdline.py'...
***   File "/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/cmdline.py", line 122
    print "Help on the %s lexer:" % cls.name
                                ^
SyntaxError: Missing parentheses in call to 'print'

Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/console.py'...
Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/filter.py'...
Listing '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/filters'...
Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/filters/__init__.py'...
Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/formatter.py'...
Listing '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters'...
Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters/__init__.py'...
Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters/_mapping.py'...
***   File "/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters/_mapping.py", line 46
    print module_name
                    ^
SyntaxError: Missing parentheses in call to 'print'

Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters/html.py'...
***   File "/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters/html.py", line 518
    except IOError, err:
                  ^
SyntaxError: invalid syntax

Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters/other.py'...
Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters/redtabhtml.py'...
Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters/rtf.py'...
Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/formatters/terminal.py'...
Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/lexer.py'...
***   File "/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/lexer.py", line 456
    except Exception, err:
                    ^
SyntaxError: invalid syntax

Listing '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/lexers'...
Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/lexers/__init__.py'...
Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/lexers/_mapping.py'...
***   File "/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/lexers/_mapping.py", line 36
    print module_name
                    ^
SyntaxError: Missing parentheses in call to 'print'

Compiling '/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/lexers/agile.py'...
***   File "/home/erian/VeloView/build/paraview/src/paraview-build/lib/site-packages/pygments/lexers/agile.py", line 340
[... 64 lines cut ...]
[  0%] Building C object VTK/Wrapping/Tools/CMakeFiles/vtkWrappingTools.dir/vtkParsePreprocess.c.o
[  0%] Building CXX object VTK/Utilities/KWSys/vtksys/CMakeFiles/vtksys.dir/FStream.cxx.o
[  0%] Linking CXX shared library ../../../lib/libvtkDICOMParser-pv5.1.so
[  0%] Building CXX object VTK/ThirdParty/alglib/CMakeFiles/vtkalglib.dir/svd.cpp.o
[  0%] Building C object VTK/Wrapping/Tools/CMakeFiles/vtkWrappingTools.dir/vtkWrap.c.o
[  0%] Building CXX object VTK/Utilities/KWSys/vtksys/CMakeFiles/vtksys.dir/SystemInformation.cxx.o
[  0%] Building C object VTK/Wrapping/Tools/CMakeFiles/vtkWrappingTools.dir/vtkWrapText.c.o
[  0%] Built target vtkDICOMParser
[  0%] Linking C static library ../../../lib/libvtkWrappingTools-pv5.1.a
[  0%] Linking CXX shared library ../../../lib/libvtkalglib-pv5.1.so
[  0%] Built target vtkWrappingTools
[  0%] Built target vtkalglib
[  0%] Linking CXX shared library ../../../../lib/libvtksys-pv5.1.so
[  0%] Built target vtksys
Makefile:127: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMake Error at /home/erian/VeloView/build/pv-paraview-build.cmake:26 (message):
  Failed!!!

CMakeFiles/paraview.dir/build.make:113: recipe for target 'paraview/src/paraview-stamp/paraview-build' failed
make[2]: *** [paraview/src/paraview-stamp/paraview-build] Error 1
CMakeFiles/Makefile2:494: recipe for target 'CMakeFiles/paraview.dir/all' failed
make[1]: *** [CMakeFiles/paraview.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

sergiocasaspastor, 2018-07-06:
I found the error. I was using python 3 instead of python 2 for building. Changing the python path corrects the error.

melvinpmanuel, 2018-11-27:
Hi,
Could you please make your repository available. I found your repository "Ros driver for Continental ARS 408 Radar" last week and currently it is not available.It would be a great help if you can upload it github.

Thanking you

vnn0804, 2019-02-17:
hey hi try 16.04 ubuntu. and build dir not required. follow the following steps

1. First, install the following dependencies using the apt-get commands
   git
   cmake-curses-gui
   build-essential
   libboost-all-dev
   libxt-dev
   libbz2-dev
   libqt4-dev
   qt4-default
   qt4-dev-tools
   zlib1g-dev
   libpcap-dev.
   export QT_SELECT=qt4
2. create project dir
   mkdir project
3. clone the veloview project
   git clone https://github.com/Kitware/VeloView.git
4. cmake build type following commands
   a) ccmake VeloView/Superbuild
   b) type c to configure
   c) page 1 for cmake
   CMAKE_BUILD_TYPE                *Release
   CMAKE_INSTALL_PREFIX            */home/icarus/project/
   ENABLE_boost                    *OFF
   ENABLE_eigen                    *OFF
   ENABLE_liblas                   *OFF
   ENABLE_paraview                 *OFF
   ENABLE_pcap                     *OFF
   ENABLE_python                   *OFF
   ENABLE_pythonqt                 *OFF
   ENABLE_qt                       *OFF
   ENABLE_veloview                 *ON
   qt_EXTRA_CONFIGURATION_OPTIONS  *

again type c (configure) and type e for exit

d) page 2 for cmake
USE_SYSTEM_boost *ON
USE_SYSTEM_pcap *ON
USE_SYSTEM_python *ON
USE_SYSTEM_qt *ON
CMAKE_BUILD_TYPE Release
CMAKE_INSTALL_PREFIX /home/icarus/project
ENABLE_veloview ON
qt_EXTRA_CONFIGURATION_OPTIONS

type c again type c and finally type g

1. run  cmake -DENABLE_veloview:BOOL=ON VeloView/Superbuild
2. run  make -j4 and make in two different terminal
3. finally open veloview application--  cd install/bin
   ./VeloView