Kitware/VeloView #58: VeloView built, unable to run

BW25, 2018-08-28:
Hello, I have built Veloview on Ubuntu Mate, but am unable to run it.

I have located the executable Veloview in this location, */VeloView/build/veloview/src/veloview-build/bin according to the directions provided here: https://github.com/Kitware/VeloView/issues/19

Attempting to run it by clicking on it in the directory states "There is no application installed for shared library files".
Attempting to run it in the terminal with ./VeloView gives a segmentation fault (core dump).

I have heard "you built using the SuperBuild, you need to explicitly activate the VeloView project with CMake."

How does one do this?

If this is not the issue, is there a problem with my build?

If anyone has any idea, please let me know.
Thanks.

bastienjacquet, 2018-08-29:
Hello,

The path you mention refer to a setup using the Superbuild.
If you correctly built the whole Superbuild, the needed libs and executables of each subproject are copied in */VeloView/build/install/bin/
[the VeloView program itself is a subproject (built in veloview/src/veloview-build) of the VeloView Superbuild which contains/build all needed dependencies]

Best,
Bastien Jacquet

etanx, 2018-11-20:
@BW25 I am also trying to get VeloView to run on a RasPi3! But am still trying through push through builts and installation.

bastienjacquet, 2018-12-14:
Hello,

VeloView on Raspberry Pi has not been tested yet. It could work, but also could not work.
But, it seems that no specific problems should arise (meaning that the process should be entirely similar to building from scratch VeloView for other linux machine).

VeloView is only a slight layer on top of Paraview, which happen to be buildable for ARM:
https://gitlab.com/arm-hpc/packages/wikis/packages/paraview

Let us know if you make any step forward on ARM.

Best,
Bastien Jacquet