Kitware/VeloView #61: Select calibration via Python API

medakk, 2018-11-16:
I'm trying to automate some conversions of pcap files, and I'm using the Python API for this. However, I'm prompted with a GUI window to select the sensor calibration everytime I run "vv.openPCAP".

I understand that this is documented in the user guide:
> However, the automation is limited by functions which will required an user interaction 
such as choosing a sensor calibration

But its very difficult to perform batch operations without this functionality.
Is it possible to add this? Perhaps as an argument to openCAP

bastienjacquet, 2018-11-19:
Hello,
We are continuously improving VeloView, so we will keep this in mind for future releases. This would be a nice feature. 
If that is a feature that you need in the near future, you can consider funding it via a support contract through Kitware.
Best regards,
Bastien Jacquet - Kitware VeloView Team

arpitpara, 2021-03-04:
In continuation of above questions. Is there any option to set calibration file in script only, so we don't have to select the calibration file from the window.?
I'm trying to give arguments in openPCAP() function itself but it's giving error that it doesn't contain any attribute like 'calibrationfilename'.