Kitware/VeloView #62: After long time run, VeloView has no display.

buaafish, 2018-11-19:
We use VeloView to display lidar pointcloud data. After long time run, about one or two days, the lidar is runing normally, but Veloview has no display. If we use ROS driver, it doesn't have this problem.

After we restart Veloview, it display good.

So we wonder if VeloView has this problem or paraview has this problem?

Do you have any suggestions about this problem?

wanjiruanjian, 2019-02-20:
We also have the same problem. Have you solved this problem?

songchaoshuai, 2019-02-26:
As mentioned  above, I also have the same problem.
Note：
1.   the  former packet number in the status bar per ten minute is less than the latter. 
Do you have any idea?
Look forward to your reply.
Thank you!