Kitware/VeloView #68: Problems in making a executable files for VeloView

jjimin, 2019-02-11:
I'm having problems in executing a VeloView.
I'm trying this on Ubuntu 18.04.1 LTS as a virtual machine on windows 10.

[https://sensors-in-robotics.readthedocs.io/en/latest/vlp-16.html](url)

I followed the steps from the link above and got the results below.

user01@ubuntu-user01:~/VeloView/build$ cmake -DENABLE_veloview:BOOL=ON ../Superbuild
-- Enabling qt since needed by: paraview;pythonqt;veloview
-- Enabling python since needed by: paraview;pythonqt
-- Enabling pythonqt since needed by: paraview
-- Enabling paraview since needed by: veloview
-- Enabling pcap since needed by: veloview
-- Enabling boost since needed by: liblas;veloview
-- Enabling eigen since needed by: veloview
-- Enabling liblas since needed by: veloview
-- Enabling veloview as requested.
-- PROJECTS_ENABLED qt;python;pythonqt;paraview;pcap;boost;eigen;liblas;veloview
-- Determined Source Version : 2.0.0-665-g9eed8c9
-- Git version (2.0.0-665-g9eed8c9) differs from version in file (3.5.0) at /home/user01/VeloView/Superbuild/../version.txt
-- Bundled package name will be: VeloView-2.0.0-665-g9eed8c9-20190211-Linux-64bit
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user01/VeloView/build

My settings in `ccmake ../Superbuild` is like this.

 CMAKE_BUILD_TYPE                 Release                                      
 CMAKE_INSTALL_PREFIX             /usr/local                                   
 ENABLE_veloview                  ON                                           
 QT_QMAKE_EXECUTABLE              /usr/bin/qmake                               
 USE_SYSTEM_boost                 ON                                           
 USE_SYSTEM_pcap                  OFF                                          
 USE_SYSTEM_python                ON                                           
 USE_SYSTEM_qt                    ON                                           
 qt_EXTRA_CONFIGURATION_OPTIONS    

But, I can't find the executable.
Is it right to find the executable after doing this?

I thought I should compile the superbuild so I also tried `cmake --build .`, but when I run it, my vm stops. 
Does anybody give me any clue? Thanks and sorry for my poor English.

vnn0804, 2019-02-12:
hey hi try 16.04 ubuntu. and build dir not required. follow the following steps
1) First, install the following dependencies using the apt-get commands
   git
cmake-curses-gui
build-essential
libboost-all-dev
libxt-dev
libbz2-dev
libqt4-dev
qt4-default
qt4-dev-tools
zlib1g-dev 
libpcap-dev.
export QT_SELECT=qt4
2) create project dir  
     mkdir project
3) clone the veloview project
    git clone https://github.com/Kitware/VeloView.git
4) cmake build type following commands 
a) ccmake VeloView/Superbuild
b) type c to configure
c) page 1 for cmake
CMAKE_BUILD_TYPE                *Release                                      
 CMAKE_INSTALL_PREFIX            */home/icarus/project/                        
 ENABLE_boost                    *OFF                                          
 ENABLE_eigen                    *OFF                                          
 ENABLE_liblas                   *OFF                                          
 ENABLE_paraview                 *OFF                                          
 ENABLE_pcap                     *OFF                                          
 ENABLE_python                   *OFF                                          
 ENABLE_pythonqt                 *OFF                                          
 ENABLE_qt                       *OFF                                          
 ENABLE_veloview                 *ON                                           
 qt_EXTRA_CONFIGURATION_OPTIONS  *                                             
 
again type c (configure) and type e for exit

d) page 2 for cmake
USE_SYSTEM_boost                *ON                                           
 USE_SYSTEM_pcap                 *ON                                           
 USE_SYSTEM_python               *ON                                           
 USE_SYSTEM_qt                   *ON                                           
 CMAKE_BUILD_TYPE                 Release                                      
 CMAKE_INSTALL_PREFIX             /home/icarus/project                         
 ENABLE_veloview                  ON                                           
 qt_EXTRA_CONFIGURATION_OPTIONS                                                

type c  again type c and finally type g

5) run  cmake -DENABLE_veloview:BOOL=ON VeloView/Superbuild

6) run  make -j4 and make in two different terminal
7) finally open veloview application--  cd install/bin
                                                            ./VeloView

jjimin, 2019-02-17:
Thanks for your kind advice.
It's very helpful to me and now I solved my problem. Thanks!

closed 2019-02-17 by jjimin