Kitware/VeloView #73: /Superbuild/common-superbuild    does not contain a CMakeLists.txt file.

vthotla88, 2019-04-03:
CMake Error at CMakeLists.txt:204 (add_subdirectory):
  The source directory

    /home/vivek/AEye/VeloViewSource/VeloView_Clion/VeloView-source/Superbuild/common-superbuild

  does not contain a CMakeLists.txt file.

-- Configuring incomplete, errors occurred!
See also "/home/vivek/AEye/VeloViewSource/VeloView_Clion/VeloView-build/CMakeFiles/CMakeOutput.log".

PierreGuilbertF, 2019-04-03:
Hello,

Did you updated the submodules using:
git submodule update --init --recursive 
?

Regards,
Pierre Guilbert

vthotla88, 2019-04-03:
Hello Pierre, 

Yes I did. The folder is still empty after that. 

Thanks,

PierreGuilbertF, 2019-04-04:
Hello,

Can you try: 
git submodule update --init Superbuild/common-superbuild
?

The problem was that without specifying the submodule to init, git tried to init the TestData module that is hosted on a public gitlab that required an account to download the sources.
It should works

Regards,
Pierre Guilbert

closed 2019-04-04 by vthotla88