Kitware/VeloView #74: fatal error: vtkSmartPointer.h: No such file or directory compilation terminated.

vthotla88, 2019-04-03:
In file included from /home/vivek/AEye/VeloViewSource/VeloView_Clion/VeloView-source/VelodyneHDL/Common/vtkCarGeometricCalibration.cxx:1:0:
/home/vivek/AEye/VeloViewSource/VeloView_Clion/VeloView-source/VelodyneHDL/Common/vtkCarGeometricCalibration.h:28:29: fatal error: vtkSmartPointer.h: No such file or directory
compilation terminated.

In file included from /home/vivek/AEye/VeloViewSource/VeloView_Clion/VeloView-source/VelodyneHDL/Common/vtkGeometricCalibration.h:71:0,
                 from /home/vivek/AEye/VeloViewSource/VeloView_Clion/VeloView-source/VelodyneHDL/Common/vtkGeometricCalibration.cxx:20:
/home/vivek/AEye/VeloViewSource/VeloView_Clion/VeloView-source/VelodyneHDL/Common/vtkTemporalTransforms.h:4:26: fatal error: vtkDataArray.h: No such file or directory
compilation terminated.

CMakeFiles/VelodyneHDL.dir/build.make:88: recipe for target 'CMakeFiles/VelodyneHDL.dir/Common/vtkCarGeometricCalibration.cxx.o' failed
make[2]: *** [CMakeFiles/VelodyneHDL.dir/Common/vtkCarGeometricCalibration.cxx.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/VelodyneHDL.dir/build.make:127: recipe for target 'CMakeFiles/VelodyneHDL.dir/Common/vtkGeometricCalibration.cxx.o' failed

make[2]: *** [CMakeFiles/VelodyneHDL.dir/Common/vtkGeometricCalibration.cxx.o] Error 1
In file included from /home/vivek/AEye/VeloViewSource/VeloView_Clion/VeloView-source/VelodyneHDL/Common/vtkConversions.cxx:18:0:
/home/vivek/AEye/VeloViewSource/VeloView_Clion/VeloView-source/VelodyneHDL/Common/vtkConversions.h:25:23: fatal error: vtkPoints.h: No such file or directory
compilation terminated.

CMakeFiles/VelodyneHDL.dir/build.make:101: recipe for target 'CMakeFiles/VelodyneHDL.dir/Common/vtkConversions.cxx.o' failed
make[2]: *** [CMakeFiles/VelodyneHDL.dir/Common/vtkConversions.cxx.o] Error 1
In file included from /home/vivek/AEye/VeloViewSource/VeloView_Clion/VeloView-source/VelodyneHDL/Common/Network/vvPacketSender.cxx:15:0:
/home/vivek/AEye/VeloViewSource/VeloView_Clion/VeloView-source/VelodyneHDL/Common/Network/vvPacketSender.h:16:31: fatal error: vtkSystemIncludes.h: No such file or directory
compilation terminated.

CMakeFiles/VelodyneHDL.dir/build.make:75: recipe for target 'CMakeFiles/VelodyneHDL.dir/Common/Network/vvPacketSender.cxx.o' failed
make[2]: *** [CMakeFiles/VelodyneHDL.dir/Common/Network/vvPacketSender.cxx.o] Error 1
/home/vivek/AEye/VeloViewSource/VeloView_Clion/VeloView-source/VelodyneHDL/Common/vtkEigenTools.cxx:24:21: fatal error: vtkMath.h: No such file or directory
compilation terminated.

CMakeFiles/VelodyneHDL.dir/build.make:114: recipe for target 'CMakeFiles/VelodyneHDL.dir/Common/vtkEigenTools.cxx.o' failed
make[2]: *** [CMakeFiles/VelodyneHDL.dir/Common/vtkEigenTools.cxx.o] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/VelodyneHDL.dir/all' failed
make[1]: *** [CMakeFiles/VelodyneHDL.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

PierreGuilbertF, 2019-04-04:
Hello,

It seems that the VeloView project build configuration does not know where to find the VTK dependencies. I guess your VTK_DIR cmake variable is empty.
Have you already build the Paraview project using the superbuild ? Or have you tried to use your system VTK / Paraview ?

Normally, the project should builder easily by following the instructions mentioned in the manual:
https://github.com/Kitware/VeloView/blob/master/Documentation/VeloView_Developer_Guide.md#linux-dependencies-

Let me know if you still have troubles to build VeloView

Regards,
Pierre Guilbert

PierreGuilbertF, 2019-04-04:
Once the issue #73 is resolved you should be able to build easily VeloView by following the Superbuild instructions

vthotla88, 2019-04-04:
Thank you so much for the help @PierreGuilbertF. I was able to resolve issue 73 and also get all the dependencies as mentioned. Now I have a new error, similar to one reported earlier but couldn't find a solution for it online. 
"[  6%] Built target python
[ 12%] Built target cxx11
[ 18%] Built target zlib
[ 20%] Performing build step for 'freetype'
libtool: compile:  gcc -pedantic -ansi -I/home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/src/src/autofit -I/home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/build -I/home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/src/builds/unix -I/home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/src/include -c -Wall -fPIC -I/home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/install/include -DFT_CONFIG_OPTION_SYSTEM_ZLIB -DFT_CONFIG_OPTION_USE_BZIP2 -I/usr/include/libpng12 -DFT_CONFIG_OPTION_USE_PNG -I/usr/include/harfbuzz -DFT_CONFIG_OPTION_USE_HARFBUZZ "-DFT_CONFIG_CONFIG_H=<ftconfig.h>" -DFT2_BUILD_LIBRARY "-DFT_CONFIG_MODULES_H=<ftmodule.h>" /home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/src/src/autofit/autofit.c  -fPIC -DPIC -o /home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/build/.libs/autofit.o
In file included from /usr/include/harfbuzz/hb-blob.h:34:0,
                 from /usr/include/harfbuzz/hb.h:31,
                 from /home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/src/src/autofit/afshaper.h:29,
                 from /home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/src/src/autofit/afglobal.h:26,
                 from /home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/src/src/autofit/afcjk.c:29,
                 from /home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/src/src/autofit/autofit.c:24:
/usr/include/harfbuzz/hb-common.h:93:29: warning: ISO C restricts enumerator values to range of ‘int’ [-Wpedantic]
 #define HB_TAG(c1,c2,c3,c4) ((hb_tag_t)((((uint8_t)(c1))<<24)|(((uint8_t)(c2))<<16)|(((uint8_t)(c3))<<8)|((uint8_t)(c4))))
                             ^
/usr/include/harfbuzz/hb-common.h:97:20: note: in expansion of macro ‘HB_TAG’
 #define HB_TAG_MAX HB_TAG(0xff,0xff,0xff,0xff)
                    ^
/usr/include/harfbuzz/hb-common.h:316:29: note: in expansion of macro ‘HB_TAG_MAX’
   _HB_SCRIPT_MAX_VALUE    = HB_TAG_MAX, /*< skip >*/
                             ^
In file included from /home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/src/src/autofit/autofit.c:35:0:
/home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/src/src/autofit/afscript.h:35:11: error: ‘HB_SCRIPT_ADLAM’ undeclared here (not in a function)
           HB_SCRIPT_ADLAM,
           ^
/home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/src/src/autofit/afshaper.c:90:38: note: in definition of macro ‘SCRIPT’
 #define SCRIPT( s, S, d, h, H, ss )  h,
                                      ^
/home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/src/src/autofit/afscript.h:278:11: error: ‘HB_SCRIPT_OSAGE’ undeclared here (not in a function)
           HB_SCRIPT_OSAGE,
           ^
/home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/src/src/autofit/afshaper.c:90:38: note: in definition of macro ‘SCRIPT’
 #define SCRIPT( s, S, d, h, H, ss )  h,
                                      ^
/home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/src/src/autofit/rules.mk:74: recipe for target '/home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/build/autofit.lo' failed
make[3]: *** [/home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/freetype/build/autofit.lo] Error 1
CMake Error at /home/vivek/AEye/VeloViewSource/VeloView/VeloView-build/common-superbuild/sb-freetype-build.cmake:47 (message):
  Failed with exit code 2

common-superbuild/CMakeFiles/freetype.dir/build.make:114: recipe for target 'common-superbuild/freetype/stamp/freetype-build' failed
make[2]: *** [common-superbuild/freetype/stamp/freetype-build] Error 1
CMakeFiles/Makefile2:509: recipe for target 'common-superbuild/CMakeFiles/freetype.dir/all' failed
make[1]: *** [common-superbuild/CMakeFiles/freetype.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2"

PierreGuilbertF, 2019-04-08:
Hello,

This error can be avoid using the option USE_SYSTEM_qt5 instead of using the superbuild qt5. We are aware of this error and it should be fixed soon.
You can download qt here: https://www.qt.io/download
You should install the qt5.10.0 version

Regards,
Pierre Guilbert

PierreGuilbertF, 2019-04-08:
PS: Once qt5.10.0 is installed you might need to indicate the location of the Qt5Config.cmake to the superbuild project, the file is located here:
${PATH_TO_QT}/Qt/5.10.0/gcc_64/lib/cmake/Qt5 (linux)
or
${PATH_TO_QT}/Qt5.10.0/5.10.0/msvc2015_64/lib/cmake/Qt5 (windows)