Kitware/VeloView #81: Various issues with fresh build (superbuild method, Windows 10)

Matioupi, 2019-04-30:
Hello,
I just achieved Veloview superbuild (boh Release and RelWithDebInfo  in 2 separate build dirs)
The build id is 3.5.0-602-g4f7351b 64 bits. Windows 10 machine.

The command line used is (I was expecting to enable the SLAM as described here  : https://github.com/Kitware/VeloView/issues/51)

cmake C:\Users\Mathieu\Documents\Devel2\Softs\VeloView-source\Superbuild -GNinja -DCMAKE_BUILD_TYPE=Release -DUSE_SYSTEM_qt5=True -DENABLE_pcl=ON -DENABLE_ceres=ON -DQt5_DIR="C:/Qt/Qt5.10.0/5.10.0/msvc2015_64/lib/cmake/Qt5"

The software is built and launch properly.
Now the issues I have faced :

- When openning a pcap file, the software crashes. (I tryed those pcap : https://data.kitware.com/#folder/5b7fff608d777f06857cb539 ). Neither the debug or release builds let me access any info about what is causing the crash.

- When hooking to a real Velodyne (Puck Hi-Res), the 3D view is displaying data, but it seems that it is refreshed at a very slow rate.

- When trying to record pcap ata from within Veloview, I receive the following error message in the Error console : 
`Traceback (most recent call last):
  File "C:\VeloView-build\install\bin\site-packages\veloview\applogic.py", line 1201, in onRecord
    recordFile(fileName)
  File "C:\VeloView-build\install\bin\site-packages\veloview\applogic.py", line 1182, in recordFile
    sensor.OutputFile = filename
  File "C:\VeloView-build\install\lib\paraview-5.4\site-packages\paraview\servermanager.py", line 456, in __setattr__
    setter(self, value)
  File "C:\VeloView-build\install\lib\paraview-5.4\site-packages\paraview\servermanager.py", line 2525, in setProperty
    return self.SetPropertyWithName(propName, value)
  File "C:\VeloView-build\install\lib\paraview-5.4\site-packages\paraview\servermanager.py", line 349, in SetPropertyWithName
    prop.SetData(arg)
  File "C:\VeloView-build\install\lib\paraview-5.4\site-packages\paraview\servermanager.py", line 813, in SetData
    self.SMProperty.SetElement(idx, val)
TypeError: SetElement argument 2: (unicode conversion error)`

About the SLAMpart, I'm not sure if it got enabled :
- When accessing the Calibration item in the Pipeline Browser view and pressing crtl+space, the pop up windows is empty (but I could only run this with live data and not from a loaded pcap)

etanx, 2019-05-01:
@Matioupi Strange, my CMakeCache.txt only has the first two. Did you get to use cmake-gui to enable pcl and ceres (see https://github.com/Kitware/VeloView/issues/51#issuecomment-487028482) after the first installation as well? After the first installation run I could see the pipeline browser but found no SLAM modules. Had to do the second installation after tweaking the cmake file.

Matioupi, 2019-05-21:
The issue in openning the .pcap files I retrieved from internet si actually related to the path were files are stored : C:\User\UserName\Téléchargement\filename.pcap
As soon as I move the file to C:\TestDir\ they do load correctly.

bastienjacquet, 2019-05-23:
Yes. Paths with non-ASCII letters are still creating trouble. That will be fixed soon.
Please avoid such paths for now.
Also: Please de not have a Windows user name with such characters (this is allowed but risky), as the application won't handle accessing its Win-assigned config folder.