Kitware/VeloView #84: Can't use the function: Save CSV (File->Save As->Save CSV)

Wilson1228, 2019-06-17:
Hi, When I use VeloView 3.5.0, it's easy to use this function to save CSV files. But after clone the source code and rebuild the project then open VeloView execution file in the path : VeloView-build\install\bin, it become fail. Would you please tell me how to solve this issue? Thanks a lot!

Wilson1228, 2019-06-17:
Console:

Traceback (most recent call last):
  File "C:\VeloView-build\install\bin\site-packages\veloview\applogic.py", line 920, in onSaveCSV
    oldTransform = transformMode()
  File "C:\VeloView-build\install\bin\site-packages\veloview\applogic.py", line 1811, in transformMode
    if reader.ApplyTransform:
  File "C:\VeloView-build\install\lib\paraview-5.4\site-packages\paraview\servermanager.py", line 485, in __getattr__
    return getattr(self.SMProxy, name)
AttributeError: 'vtkPVServerManagerCorePython.vtkSMSourceProxy' object has no attribute 'ApplyTransform'

PierreGuilbertF, 2019-06-25:
Hello @Wilson1228,

Yes, the export as csv feature is broken when launched from the GUI interface, this is something that will be fixed on the next release.
However, you can still save your data as csv using `ctrl + s` and selecting .csv export file format.

Regards,
Pierre Guilbert

Wilson1228, 2019-06-26:
Great!Thank you for your help!

Wilson1228, 2019-07-05:
Hi @PierreGuilbertF ,
Could you tell me when to release the next official version of VeloView? It's seems that released a new version for a log time in 2017.

PierreGuilbertF, 2019-07-05:
We should be able to release a new version next September.
The new version will include bug-fixes and new features to manage Lidar-Camera system with tools to perform semi-automatic calibration. The slam algorithm should also include a loop-closure routine if I finish it in time :)