Kitware/VeloView #85: Memory Leak - Windows 10

Chris1990, 2019-06-19:
Followed build instructions and managed to compile/run VeloView in Windows 10. 

I have been using the record feature to record lidar data from a Velodyne Puck when I noticed there is a memory leak. First I thought it was related to the recording feature so I tested with just connecting the puck and leaving it be. 

The screenshot below is the application running just for 6 minutes, as I write this I can see the memory rising.  

[image]

Any thoughts? 

Cheers,
Chris

PierreGuilbertF, 2019-06-25:
Hello @Chris1990,

Thank you very much for this detailed bug report.
I will investigate on it.
Right now, my first guess would be that we have an internal queue that is enqueuing the last N-received packets (we settled N to 100 is I remember right). Thus, at the beginning of a recording the memory is expected to increase as long as the queue is not full.

Regards,
Pierre Guilbert

Chris1990, 2019-06-25:
Hello @PierreGuilbertF 

That would make sense, the memory did increase while I wasn't recording though. I left it for maybe a hour or so before I had to move onto something else. 

Let me know if you need any more information or data. 

Cheers,
Chris

alvin198811, 2020-08-13:
Hello @Chris1990   when compiling veloview, QT version must be 5.10? whether other version can work or not?thanks.

alvin198811, 2020-08-13:
Hello @Chris1990   when compiling veloview, QT version must be 5.10? whether other version can work or not?thanks.

aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa