Kitware/VeloView #87: VeloView cannot open VLP-32C pcap files: error VLP-32c.xml not found

RBarron01, 2019-06-24:
I am working building VeloView from source following[ these instructions](https://github.com/Kitware/VeloView/blob/master/Documentation/VeloView_Developer_Guide.md#superbuild-overview) for Linux (running Ubunut 16.04). I have multiple .pcap files that I have collected and tested on another basic build (not building from source) and they are fine but when I open them on this build I get the error:
`~/lidar/VeloView-build/common-superbuild/veloview/build/bin/../../share/VLP-32c.xml: File not found!
In ~/lidar/VeloView-source/VelodyneHDL/IO/Lidar/Common/vtkLidarProvider.cxx, line 54
vtkLidarReader (0x409f0f0)`
After that I have a lot of other errors but they all seem to be pointing back to the first one since they all have to do with the file not being able to be opened.

I'm trying to enable the SLAM capability (referencing [Issue #51](https://github.com/Kitware/VeloView/issues/51)). I have pcl and ceres enabled in the cmake gui and while compiling.

PierreGuilbertF, 2019-06-25:
Hello @RBarron01,

This is weird since we are providing the .xml calibration file for: HDL-32, Pucj Hi-Res, Puck LITE, VLP-16 and VLP-32C.
Have you run `make install` after the project have been built ?
The install step is the one in charge to copy the calibration files from the VeloView sources to the correct install/share folder.
Once you have run `make install`, if it still doesn't work you can manually add the calibration file. To do that you should click on the `Add` button on the configuration dialog and then select the correct file that should be located in the folder: `${PATH_TO_VELOVIEW_SOURCE}/share/VLP-32c.xml`

[image]

Regards,
Pierre Guilbert

RBarron01, 2019-06-25:
Thank you! Yes, I have run make install successfully. Adding the configuration file manually worked.
Thanks again!

closed 2019-06-28 by RBarron01