Kitware/VeloView #102: Errors in Installing VeloView on Ubuntu 18.04

zeye-shen, 2020-04-22:
Hi,
 there is my step to build veloview.
1. mkdir veloview
2.cd veloview
3.git clone https://github.com/Kitware/VeloView.git Veloview-source
4. cd Veloview-source
5. git submodule update --init Superbuild/common-superbuild
6. cd ..
7. mkdir VeloView-build
8.cd Veloview-build
9.cmake ../Veloview-source/Superbuild -DCMAKE_BUILD_TYPE=Release
10. make -j6
there is error for last step.
zeyeshen@zeyeshen-desktop:~/Documents/veloview/VeloView-build$ make -j6
[  6%] Built target zlib
[ 12%] Built target gperf
[ 18%] Built target python
[ 18%] Performing configure step for 'libxml2'
[ 24%] Built target cxx11
[ 24%] Performing configure step for 'pcap'
[ 26%] Performing update step for 'eigen'
[ 27%] Built target cxx11_check
[ 33%] Built target freetype
[ 35%] Performing build step for 'png'
make[3]: warning: -jN forced in submake: disabling jobserver mode.
[image]
[image]
[image]
[image]
[image]

[ 40%] Built target boost
[ 41%] Performing configure step for 'eigen'
[ 49%] Built target liblas
[  5%] Built target png-fix-itxt
[ 40%] Built target genfiles
Not searching for unused variables given on the command line.
abort: repository /home/zeyeshen/Documents/veloview/VeloView-build/common-superbuild/eigen/src not found!
[ 94%] Built target png
abort: repository /home/zeyeshen/Documents/veloview/VeloView-build/common-superbuild/eigen/src not found!
-- Standard libraries to link to explicitly: none
[ 97%] Linking C executable pngfix
checking build system type... libpng16.so.16.23.0: undefined reference to `png_init_filter_functions_neon'
collect2: error: ld returned 1 exit status
CMakeFiles/pngfix.dir/build.make:97: recipe for target 'pngfix' failed
make[5]: *** [pngfix] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/pngfix.dir/all' failed
make[4]: *** [CMakeFiles/pngfix.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMake Error at /home/zeyeshen/Documents/veloview/VeloView-build/common-superbuild/sb-png-build.cmake:47 (message):
  Failed with exit code 2

common-superbuild/CMakeFiles/png.dir/build.make:114: recipe for target 'common-superbuild/png/stamp/png-build' failed
make[2]: *** [common-superbuild/png/stamp/png-build] Error 1
CMakeFiles/Makefile2:1450: recipe for target 'common-superbuild/CMakeFiles/png.dir/all' failed
make[1]: *** [common-superbuild/CMakeFiles/png.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
qmake: could not exec '/usr/lib/aarch64-linux-gnu/qt4/bin/qmake': No such file or directory
checking build system type... -- Found unsuitable Qt version "" from NOTFOUND
./config.guess: unable to guess system type

This script, last modified 2009-11-20, has failed to recognize
the operating system you are using. It is advised that you
download the most up to date version of the config scripts from

  http://git.savannah.gnu.org/gitweb/?p=config.git;a=blob_plain;f=config.guess;hb=HEAD
and
  http://git.savannah.gnu.org/gitweb/?p=config.git;a=blob_plain;f=config.sub;hb=HEAD

[... 503 lines cut ...]
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/NonLinearOptimization/r1mpyq.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/NonLinearOptimization/r1updt.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/NonLinearOptimization/rwupdt.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/NumericalDiff/NumericalDiff.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/Polynomials/Companion.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/Polynomials/PolynomialSolver.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/Polynomials/PolynomialUtils.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/Skyline/SkylineInplaceLU.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/Skyline/SkylineMatrix.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/Skyline/SkylineMatrixBase.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/Skyline/SkylineProduct.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/Skyline/SkylineStorage.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/Skyline/SkylineUtil.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/SparseExtra/BlockOfDynamicSparseMatrix.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/SparseExtra/DynamicSparseMatrix.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/SparseExtra/MarketIO.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/SparseExtra/MatrixMarketIterator.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/SparseExtra/RandomSetter.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/Splines/Spline.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/Splines/SplineFitting.h
-- Up-to-date: /home/zeyeshen/Documents/veloview/VeloView-build/install/include/eigen3/unsupported/Eigen/src/Splines/SplineFwd.h
[ 50%] Completed 'eigen'
[ 52%] Built target eigen
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

zeye-shen, 2020-04-22:
the second picture should be the first

khaledgabr77, 2020-08-04:
i made this steps for installation and it's working .
1- mkdir source_veloview
2- cd source_veloview
3- git clone https://github.com/Kitware/VeloView.git Veloview-source
3- cd Veloview_source
4- git submodule update --init Superbuild/common-superbuild
5- cd ../..
6- mkdir build_VeloView
7- cd build_VeloView
8- cmake ../source_veloview/Veloview-source/Superbuild -DCMAKE_BUILD_TYPE=Release
9- make -j6 
10- cd common-superbuild/veloview/build
11- make insatll

AndreV84, 2020-08-19:
it fails at the step9