Kitware/VeloView #108: Superbuild won't build on nvidia Jetson Xavier AGX

AndreV84, 2020-08-17:
```
common.copy /home/agx/VeloView/Superbuild/install/include/boost/tr1/tr1/valarray
common.copy /home/agx/VeloView/Superbuild/install/include/boost/tr1/tr1/locale
...updated 13390 targets...
[ 43%] Completed 'boost'
[ 43%] Built target boost
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2

```

AndreV84, 2020-08-17:
```
epends.cmake
-- Installing: /home/agx/VeloView/Superbuild/install/share/cmake/libLAS/liblas-depends-release.cmake
[ 55%] Completed 'liblas'
[ 55%] Built target liblas
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2

```

Rotoslider, 2021-01-31:
were you ever able to get this to build on the AGX. I have the same error as your first one on my AGX. 

```
[ 43%] Built target boost
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2
```

MatthewOliverKing, 2021-08-21:
I get a similar error, but while trying to build eigen, but on my PC (Ubuntu 20.04) and Jetson Nano (18.04). Did either of you manage to figure this one out?